OrcaSlicer/OrcaSlicer #2203: Blank Setup Wizard, "No provider of eglCreateImage found. Requires one of: EGL 15"

SeongGino, 2023-09-23:
### OrcaSlicer Version

1.7.0

### OS version

Arch Linux

### Additional system information

CPU: AMD Ryzen 5 5600X
RAM: 16GB DDR4
GPU: NVIDIA GeForce RTX 3060ti (w/ prop drivers 535.104.5)

### Printer

N/A

### How to reproduce

1. Run Orca-slicer appimage from the terminal.
2. Observe the error in the terminal output, and the blank Setup Wizard.

### Actual results

Orca Slicer takes a while to startup, and when it does, the initial Setup Wizard is blank and unresponsive, rendering the app unusable; close window requests don't work, the application has to be KILLed/interrupted to exit.
[image]

(NVIDIA card works fine with OpenGL/Vulkan/OpenCL hardware acceleration in other applications and games, including PrusaSlicer, FWIW)

### Expected results

Orca Slicer should work?

### Project file & Debug log uploads

[debug_Fri_Sep_22_21_55_21_42577.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/12705376/debug_Fri_Sep_22_21_55_21_42577.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

AdeilnAcinrst, 2023-10-11:
Same thing here.

Tried to compile from source and run disabling features listed:

LIBGL_DRI2_DISABLE=1 LIBGL_DRI3_DISABLE=1 ./OrcaSlicer_ubu64.AppImage

Same blank screen.

Debian AMD Ryzen 9, 64GB NVidia GTX 1070

Hitting ESC allows me to enter the application, but it's not useful because I'm not able to change settings, almost all components are "readonly"

HakunMatat4, 2023-12-16:
@astrolemonade , interesting the solution to disable Nvidia.

I'm on Linux Mint Cinnamon ( Ubuntu based but without the dramas and more like Debian (stable , light, etc ).

Lenovo Legion laptop, Ryzen7, RTX 3050Ti in Performance Mode enabled.
Intel mode makes anything 3D to have terrible lag on external monitors.

Those still having problems, I suggest you visit BambuLab GitHub ticket issue number 12 about Linux version of the slicer.
There is where it all started and you are more likely to find a solution/better support there.

We only have the AppImage now coz of those folks.

AByzhynar, 2024-02-13:
The same issue on Ubuntu 22.04 with OrccaSlicer 1.8.1 and 1.9.0

katodo, 2024-03-07:
same problem for me on Ubuntu 22.04 with Nvidia and last Orcaslicer revision ( OrcaSlicer_Linux_V1.9.1 Appimage)

closed 2024-06-13 by github-actions[bot]

SeongGino, 2024-06-14:
Update because of staleness (thanks bot): issue does not seem to be present in recent Orca Slicers, at least as of the 2.0 bin and NVIDIA 550.

spusuf, 2024-06-27:
The issue is still present with OrcaSlicer 2.1.1 with NVIDIA 555. However, ``WEBKIT_DISABLE_DMABUF_RENDERER=1 ./OrcaSlicer_Linux_V1.9.1.AppImage`" does work. Unfortunately it seems to be an issue with BambuStudio upstream as there are a few issues on their github page.

spusuf, 2024-06-27:
Just wanted to add if you're using Gear Lever to add Orca Slicer to your DE's application list , you can add an environment variable with key ```WEBKIT_DISABLE_DMABUF_RENDERER``` and value ```1``` . It would be the same for people making .desktop files, this seems like a complete workaround, even have webcam in LAN only mode on my P1S, which I can't get on Ubuntu with BambuStudio.

459below, 2024-10-18:
> Just wanted to add if you're using Gear Lever to add Orca Slicer to your DE's application list , you can add an environment variable with key `WEBKIT_DISABLE_DMABUF_RENDERER` and value `1` . It [...]

`WEBKIT_DISABLE_DMABUF_RENDERER=1 ./OrcaSlicer_Linux_V2.1.1.AppImage`

Thanks. It works for me now on Fedora 40 Silverblue.

Running normally it errors out like this:

```
/tmp/.mount_OrcaSlwEUCQ0/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```

But running it inside a `toolbox` container and using your suggested environment variable like this, made it work:

```
toolbox enter
sudo dnf install fuse fuse-libs mesa-libGL mesa-libEGL gtk3 webkit2gtk3
WEBKIT_DISABLE_DMABUF_RENDERER=1 ./OrcaSlicer.AppImage
```

bacher09, 2025-01-31:
Hey, for quite some time I was using the same workaround with `WEBKIT_DISABLE_DMABUF_RENDERER` but after WebKit update it stopped working for me (it was causing Orca to crash in a few seconds after launch). Although, I found another workaround `WEBKIT_DMABUF_RENDERER_FORCE_SHM=1`, which fixed it for me. For those who are interested, [this check appears](https://github.com/WebKit/WebKit/blob/wpewebkit-2.46.5/Source/WebKit/UIProcess/gtk/AcceleratedBackingStoreDMABuf.cpp#L86) in the same file just a little bit below the previous environment variable.

Without this environment variable (`WEBKIT_DMABUF_RENDERER_FORCE_SHM=1`), Orca crashes in WebKit with null pointer dereference:

```
orcaslicer_main[205455]: segfault at 0 ip 00007f15e1109326 sp 00007ffd88728e08 error 4 in libwebkit2gtk-4.0.so.37.71.7[7f15e0a4e000+3582000] likely on CPU 15 (core 15, socket 0)
```