OrcaSlicer/OrcaSlicer #2205: Add Per-Feature Pressure Advance

gdsolar, 2023-09-23:
**Is your feature request related to a problem? Please describe.**
Not related to a problem, but per-feature Pressure Advance is one of the most powerful tuning settings by far. Different speeds and accelerations have different optimal PA values, and this would open up another level of print quality tuning for the user.

**Which printers will be beneficial to this feature**
Every single printer ever made.

**Describe the solution you'd like**
I would like to see a menu in the filament settings that would allow the user to set Pressure Advance for specific features (inner wall, outer wall, infill, top surface, etc.) for each filament. 

**Describe alternatives you've considered**
I haven't really considered any alternative since it's a fairly simple idea that no other major slicer has.

**Additional context**
N/A

gbyleveldt, 2023-09-25:
This would be an awesome feature. I was just talking to someone else about this very feature missing in every slicer. Apparently you can mitigate the large difference between perimeters PA values by increasing square corner velocity (within limits). This means that the toolhead won’t need to slow down that much on outer perimeters for example, letting you get away with lower PA values overall. But being able to give a PA value (or ratio) for a given perimeter type would really be awesome

yet-another-average-joe, 2023-09-30:
@gdsolar : 

Cura : the Klipper plugin adds this functionality. Ok, technically, it isn't a Cura feature !

Per feature pressure advance control is the first thing I searched for, after I installed Orca. And sadly the very reason why I can't use it. Would be downgrading my toolchain, and throwing away all the hard calibration work (not speaking of print quality)...

More granularity is required : per feature speeds, flows, accelerations and PA.

armitage61, 2023-11-06:
Yes, that would be nice. 
Like that...
[image]

gdsolar, 2023-11-06:
@armitage61 Exactly like that! Is that the klipper plugin for Cura that yet-another-average-joe mentioned above?

armitage61, 2023-11-06:
Yes, that's the plugin for CURA. 
I want to have it in ORCA

closed 2024-02-20 by github-actions[bot]