OrcaSlicer/OrcaSlicer #2211: Generic Filaments not accessible

Googliola, 2023-09-23:
### OrcaSlicer Version

1.7

### OS version

Windows 10

### Additional system information

Due to the structure and dependencies of the profiles, it is not possible to add generic filaments to a specific printer. 

What is the procedure to add new filaments such as PCTG or TPU if not already contained in the resources of that specific printer?
 
[image]

### Printer

Snapmaker J1 (any other as well)

### How to reproduce

Add Printer Snapmaker J1
Add Generic PCTG filament - you cant

There is a messy solution: edit user json files

### Actual results

-

### Expected results

-

### Project file & Debug log uploads

No logs required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kylek29, 2023-09-23:
Did this work before 1.7? I'm only ask because I know BambuStudio introduced some form of distributed profiles they push occasionally to BS, be curious if this broke/altered something.

I think they're using the data from all their connected printers to build out a catalog of vendors/types.

Googliola, 2023-09-24:
As I only recently started using OrcaSlicer, I don't know if it worked before 1.7.

closed 2023-12-31 by github-actions[bot]

DigitArier, 2024-10-19:
PCTG is not yet integrated. Perhaps there should be an option to enter 'Other' or 'User' for 'Filament Type' with subsequent input (just like the option above)