OrcaSlicer/OrcaSlicer #2221: Overhang Slowdown logic somehow getting capped by Filament cooling ?

psiberfunk, 2023-09-24:
### OrcaSlicer Version

1.7.0 release

### OS version

Windows 11 / Win64

### Additional system information

N/A - Slicer logic issue

### Printer

Bambu X1C

### How to reproduce

1. Slice attached project file with settings included, observe that for layer 57 (see screenshots below) the speed for the overhang in the screenshot is mysterious much slower (15 mm/sec) than it should be.
. 

### Actual results

1. Notice that the overhand speed in the screenshot here is 15, when at minimum it should be the lesser of 60 OR the VFR-limited speed (which is ~44 mm/sec at this line width).
2. Notice also that the filament min-slowdown for cooling speed is also 15 (appears to be hitting this limit somehow, but not near the min layer speed).
3.  Notice how wh
[image]
[image]
[image]
en the slowdown-for-overhangs is turned OFF, this same segment is at speed 44 instead of 15.

### Expected results

The speed for these overhangs should NOT be governed by the min-speed set in the filament cooling tab here, because 

1. the layer time is not anywhere near the min layer time (6 seconds, this layer is ~10 sec) 
2. The min speed for the appropriate overhang is set to 60 (way above 15).
3. Turning slow-down-for overhangs OFF fixes the problem by restoring the speed to 44 mm/sec - indicating that this slowdown is happening BECAUSE of slowdown-for-overhangs.. but the wrong logic is getting applied. 
4. The correct speed here, according to the settings ought to be 44 mm/sec , since this is the lesser of Overhang slow down speed v.s. VFR-limited-speed.

### Project file & Debug log uploads

 [OverhangSlowdownLogicIncorrect.zip](https://github.com/SoftFever/OrcaSlicer/files/12709838/OverhangSlowdownLogicIncorrect.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

psiberfunk, 2023-09-25:
Log file not included because this is a reproducible slicer logic bug and not a crash or UX issue or similar.

kevinbrouwers, 2023-09-27:
Confirmed. I spent several hours trying to figure out what setting was slowing my speed down to 10mm/s when it should have been 100 mm/s. It was this bug. Thanks for posting this.

closed 2024-01-04 by github-actions[bot]