OrcaSlicer/OrcaSlicer #2227: Bridged Holes Encroachment Setting

Lt-Kammello, 2023-09-25:
Connection issues on brudged holes.

When you have an hole in a bridged surface, the perimeters are partially treated like bridges, if you choose thick bridge or not they are printed accordingly.

The issue is the amount of "encroaching" between bridged infill and hole's walls [See Pics Below].
While bridges attach to the main body walls with large amount of encroaching ensuring connection, the same does not apply for the walls of the inner hole, resulting in no connection.

I think this would be a fairly big step forward having the right amount of connection or at least an option to set it.

[image]
[image]

mpsnp, 2023-09-30:
Guess these kind of issues can be easily solved with [this modelling trick](https://youtu.be/W8FbHTcB05w?si=nmwPoyhqK_hop9Ji&t=531)

Lt-Kammello, 2023-10-10:
> Guess these kind of issues can be easily solved with [this modelling trick](https://youtu.be/W8FbHTcB05w?si=nmwPoyhqK_hop9Ji&t=531)

Sure thing but that’s not optimal.
Also bridged walls are not treated like bridges creating 2 main issues:
1) inconsistent adhesion to support due to the difference in applied flow rate;
2) lack of adhesion between bridge and wall like i showed in my pics

i will label this as a bug and i’m asking for @SoftFever attention since i think this bug is very relevant.

kindly

Lorenzo

closed 2024-01-17 by github-actions[bot]