OrcaSlicer/OrcaSlicer #2228: Crashes when slicing using auto tree supports

MacBoyPro98, 2023-09-25:
### OrcaSlicer Version

1.7.0

### OS version

macOS 13.5.1

### Additional system information

CPU: Apple M2 Max
Memory: 32 GB

### Printer

Voron 2.4

### How to reproduce

1. Open file that needs supports
2. Enable supports
3. Select "Auto Tree"
4. Click Slice Plate
5. Crash

### Actual results

The application crashes

### Expected results

The plate should be sliced

### Project file & Debug log uploads

[debug_Mon_Sep_25_09_20_47_72083.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/12716983/debug_Mon_Sep_25_09_20_47_72083.log.0.zip)
[Tron Disc.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12716993/Tron.Disc.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-09-26:
This has been fixed in master branch and will be in the release.

closed 2023-10-02 by MacBoyPro98

David-Zimm, 2024-04-30:
Commenting on this in hopes that this re opens, as im currently experiencing this issue and not sure why or a work around.
Orca 2.0.0
Windows 11 Pro
64GB Ram
CPU is an Intel 9 1400K