OrcaSlicer/OrcaSlicer #2230: Change filament feature (M600) not working with non bambulab printers

Luigisvc, 2023-09-26:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11

### Additional system information

CPU: I7 8th 8750H
16GB RAM
GPU 1050Ti 4GB

### Printer

BIQU B1 modified with klipper and H2V2S extruder

### How to reproduce

Add a stl
add a second filament in tab
slice and select the layer you want to change filament/color
re slice
export gcode to your computer and open with text editor.
Inspect Gcode and find M600 and T0 in the beginning after the Start Print macro call out
keep finding and will encounter a second M600 and T1 in the correct layer.

[image]

### Actual results

the result is that after the purge line when the printer should start printing with the loaded filament instead, it executes the macro to replace filament which consists in park the head, unload the filament and pauses waiting to the new filament. 
Also it selects T0 as extruder. Then change to T1 at the correct layer which cause to stop extruding

[image]

[image]

### Expected results

The expected behavior is to start printing with the current filament and pauses with M600 only in the desired layer as planned.
AND no changes in the extruder selection

### Project file & Debug log uploads

[BUGM600_Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/12721041/BUGM600_Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

wmariz, 2023-10-06:
As a workaround, put:

`SAVE_VARIABLE VARIABLE=ignorefc VALUE=1`

on the last line off Machine start G-code and warp the M600 macro with:

```
[gcode_macro M600]
gcode:
  {% if printer.save_variables.variables.ignorefc|float == 0 %}

  {% endif %}
  SAVE_VARIABLE VARIABLE=ignorefc VALUE=0

```

SoftFever, 2023-10-06:
Just to clarify, [M600](https://marlinfw.org/docs/gcode/M600.html) is a standard Filament Change gcode command recognized by most firmwares like Marlin, RRF, Prusa, and others.

Klipper doesn't come with an in-built filament change command, so we'll need to create a macro for this purpose ourselves.

Luigisvc, 2023-10-06:
> Just to clarify, [M600](https://marlinfw.org/docs/gcode/M600.html) is a standard Filament Change gcode command recognized by most firmwares like Marlin, RRF, Prusa, and others.
>
> Klipper doesn't come [...]

Yes, The macro emulating M600 I have already done in Klipper. 

As a workaround I am using the feature in Orca to add Custom Gcode and put the M600 there, but would be good if we can use the built in change filament feature without having a MMU, AMS or other multi material solution.

Luigisvc, 2023-10-06:
> As a workaround, put:
> 
> `SAVE_VARIABLE VARIABLE=ignorefc VALUE=1`
> 
> on the last line off Machine start G-code and warp the M600 macro with:
> 
> ```
> [gcode_macro M600]
> gcode:
>   {% if printer.save_variables.variables.ignorefc|float == 0 %}
> 
>   {% endif %}
>   SAVE_VARIABLE VARIABLE=ignorefc VALUE=0
> ```

I have to try this.

Ocraftyone, 2023-10-06:
@SoftFever we may be able to use the SEMM flag in the printer config and if that flag is there, don't write the first change filament

wmariz, 2023-10-06:
In my understanding, the problem described here is not the need for a macro for the M600, but rather the fact that the slicer inserts an apparently incorrect filament change command at the beginning of the gcode, right after the initialization code . My suggestion is just a temporary workaround while the bug persists.
I chose this approach because this way it is possible to use filaments that require different temperatures and settings, which is not so simple when adding an M600 manually.

wmariz, 2023-10-06:
> SEMM flag

How to do this? Can you please clarify? In my printer settings I cannot uncheck the "Single Extruder Multi Material" option. Maybe this is related to the problem?

Ocraftyone, 2023-10-06:
@wmariz sorry for not being a bit more descriptive. this would have to be a change made to the code of the program in the next update. in the mean time, i would suggest using the workaround macro @Luigisvc provided

closed 2023-10-22 by SoftFever

Ltek, 2023-12-15:
Orca 1.8.1 still pauses after purge, before Layer 1 starts

Ludancv, 2024-02-01:
> note to others find this... you MUST enable the 'Manual Change Option' under Printer Settings 'Multimaterial'
Thank you so much, that bug was driving me crazy.

Ocraftyone, 2024-02-02:
> > note to others find this... you MUST enable the 'Manual Change Option' under Printer Settings 'Multimaterial'
> 
> Thank you so much, that bug was driving me crazy.

This is not a bug. It is intended to make sure that the correct filament is loaded on automatic tool changers. It should only skip this if the user is manually changing the filaments.

Ltek, 2024-02-02:
> This is not a bug. It is intended to make sure that the correct filament is loaded on automatic tool changers. It should only skip this if the user is manually changing the filaments.

technically not a bug but its not good, and somewhat annoying, that its hidden... Orca allows you to configure filament changes in 'Prepare' while -it- knows you dont have that setting set. It should -at least- alert you to that setting not being set when you add your first filament in Prepare. Ideally, it turns it on for you after asking if you want it on.

Ocraftyone, 2024-02-03:
I don't particularly think that a popup every single time a second filament is added is the best idea. I don't feel as that would fit the current design philosophy of the UI (where it stays out of the way and doesn't typically recommend anything to the user).

I think that the solutions that should be considered are either updating printer profiles, add it to the tips section, adding it to the add printer profile dialog, and/or improving the GitHub wiki for Orca. 

I am personally leaning more towards updating the wiki and adding it as a tip.

Ltek, 2024-02-03:
> I don't particularly think that a popup every single time a second filament is added is the best idea. I don't feel as that would fit the current design philosophy of the UI (where it stays out of [...]

Orca without a doubt should ask user if they want the printer setting turned on... the the 'exact' same thing Orca does for Vase Mode... its that important... and 100% aligned with the rest of the Orca UX.

without this popup, Orca is 'staying out of the way' so much its detrimental to the UX in a big way

Allowing Orca's UI to add a filament without the user knowing (aka, being warned) that it may literally do nothing for them, is problematic in many ways and 100% bad UX for this setting

PR #2379 (2023-10-22): Add "Manual filament change" option and logic
This option does a few things that make a SEMM setup where you are manually swapping the filament on pauses a bit easier

1) disables calling the "on filament change" gcode during setup. This may be useful on a machine that can change the filament itself, but for manually changing the filament, it is an unnecessary pause. It should be assumed the first inserted filament is the first one to be used.

2) tool change command is changed to a gcode tag. This prevents errors on machines that do not understand the command possibly resulting in a failed pr [... cut]