OrcaSlicer/OrcaSlicer #2231: Bridges being sliced as Overhang wall

blownupp, 2023-09-26:
### OrcaSlicer Version

1.7.0-beta

### OS version

Windows 10.0.19045

### Additional system information

CPU: 13th Gen Intel Core i5-13600k
RAM: Corsair 64GB
GPU: EVGA 3080TI

### Printer

Voron 2.4

### How to reproduce

Slice [this bridge test:](https://www.printables.com/model/109216-bridge-test-fast-low-cost-ender-3/files) - checking/unchecking Slow Down For Overhang/Classic Mode doesn't change the result.

### Actual results

Bridged gaps are sliced as overhangs:

[image]

[image]

### Expected results

The bridged perimeters should be sliced with "Bridge flow", "Bridge density" and "Thick bridges" enforced per slicer settings.

### Project file & Debug log uploads

Attached is the STL and a saved project file with the issue prevalent.

[bridge_test_orca_errors.zip](https://github.com/SoftFever/OrcaSlicer/files/12721738/bridge_test_orca_errors.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Lt-Kammello, 2023-09-26:
Might my request/issue have to do with this?

https://github.com/SoftFever/OrcaSlicer/issues/2227#issue-1911382256

floorcat, 2023-10-09:
(OrcaSlicer 1.7.0 on linux)

I think this is the same issue here.

Benchy inner ceiling sliced wrong. (classic and arachne same result)
[image]

[image]

Unchecking "Detect overhang walls" seems to fix it.
[image]

floorcat, 2023-11-12:
still happening as of 1.8.0-rc2

blownupp, 2023-11-21:
Still happening on official 1.8 release (and on upstream Bambu Studio - you can't print overhang walls on thin air!)

Ltek, 2023-12-31:
@SoftFever this been ignored for 2 versions... its causes huge problems and it seems there is no workaround. I have mini figures that have 6 to 10 mm bridges that fail because Orca does not treat them as a bridge. Please fix.

SoftFever, 2024-01-01:
Hey guys,

There are different questions and cases involved in this thread.
I will reply to the first one here.
The reason it treated as bridge perimeter instead of bridge infill is because it's a narrow bridge and you have specified 4 wall loops, so no bridge infill is generated.
If you change the wall loops to 2, you will be able to see it:
[image]

Ltek, 2024-01-01:
@SoftFever what if Orca had the option like "Make overhang wall printable" -- the current "Make Overhang Printable" doesnt work at all for the model attached; it will plug the bottom hole completely with infill. In other slicers I've see this add a bridge or two under overhang wall, against the current solid wall to help it cantilever out.

I'm seeing the largest issue on miniature models and/or areas where the thickness changes from high to lower (area that can sustain 3,4,5 walls to and area that can only sustain 2. Orca is making that transition using Overhang Wall, and they are failing every time for me. These are only 4mm long + 10mms + 100% cooling.  Bridging is nearly perfect on that same filament doing the orca temp tower -- I did the temp tower after about 6 failed Overhang Wall attempts.

I've attached my STL so you can see it for yourself.

I modeled multiple version and printed ~20 various ones and the problem is always Overhang Walls fail (I get filament strands falling).  I had to completely redesign my part so I could avoid overhang walls and make the entire model sustain 3 walls to avoid the transition to 2 walls. If someone wants to add strength to an STL they inherited (did not design themselves), the print will fail (or at least in part) every time if it does a transition like this.

[image]
[image]
turned Precise Wall on...
[image]
[issue.zip](https://github.com/SoftFever/OrcaSlicer/files/13805564/issue.zip)

Ltek, 2024-01-01:
playing more today Now with this newer model, it creates a full overhang wall "layer"
[image]

project file attached
[Figure_FullOverhang Wall.zip](https://github.com/SoftFever/OrcaSlicer/files/13806068/Figure_FullOverhang.Wall.zip)

Ltek, 2024-01-02:
Here's one where there are more walls below than above but it still adds an Overhang Wall where its not needed at all, or a bridge would have been better. I've tried printing this a few times, 100% fan and it fails every time. 

It adds the Overhang Wall even I change to 0% infill for the model.

3 layers in order bottom to top...

[image]
[image]
[image]
[Figure_topple_MyKlipper_Hyper_30-MyKlippe__Tapered-test1.zip](https://github.com/SoftFever/OrcaSlicer/files/13807149/Figure_topple_MyKlipper_Hyper_30-MyKlippe__Tapered-test1.zip)

SadBones, 2024-01-03:
Bump. Walls that are part of bridges print at overhang wall speeds rather than bridge speed causing the print to fail.

[image]

SadBones, 2024-01-03:
> Hey guys,
>
> There are different questions and cases involved in this thread. I will reply to the first one here. The reason it treated as bridge perimeter instead of bridge infill is because it's a [...]

"bridge perimeter" is not an editable speed. intuitively, the "bridge speed" would set the bridge perimeter speed, but in practice it only sets the bridge infill speed, leaving the "bridge perimeter" speed incredibly low.

SadBones, 2024-01-03:
> Hey guys,
>
> There are different questions and cases involved in this thread. I will reply to the first one here. The reason it treated as bridge perimeter instead of bridge infill is because it's a [...]

The problem is the "bridge perimeters" are being sliced at the "overhang 75% - 100%" speed, rather than the "bridge" speed.
You would want a 90% over hang to be printed extremely slow, but the perimeters of a bridge need to be printed quickly.

YanceyA, 2024-01-08:
I see similar behavior on a model. With detect overhang walls OFF I see the expected bridging behavior in certain areas. I also see overhang walls being printed at the correct slowdown speeds depending on the overhang % of the wall. Here is a picture of some keys areas of the model showing expected bridging.
[image]

With detect overhang walls ON, some of the bridge areas are turning into high density overhang walls. This is not desired behavior and in most of my models will cause print issues. Note that this only affects some areas as the tops of holes still have perimeters. 
[image]

Here is my project file:
[Xol_Sherpa_Mount_Bridge_Overhang_Walls.zip](https://github.com/SoftFever/OrcaSlicer/files/13866475/Xol_Sherpa_Mount_Bridge_Overhang_Walls.zip)

My desired behavior would be for the bridge areas to be generated per the first photo with detect overhang walls ON. I would like to bridge and be able to reverse overhang walls and apply proper slowdowns on overhang walls (as noted above, overhang walls appears to be properly slowing down, but I see some possible odd behavior in how the slowdowns are applied).

Ltek, 2024-01-10:
> The problem is the "bridge perimeters" are being sliced at the "overhang 75% - 100%" speed, rather than the "bridge" speed. You would want a 90% over hang to be printed extremely slow, but the [...]

[image]

riddlemd, 2024-01-26:
I had this bug get me good today, was printing a case and the part of the wall that the slicer thought was bridge was not being printed before the bridge was started so the bridge just fell off into air. Turning off "Detect overhang walls" fixed it.

Ltek, 2024-01-30:
> I had this bug get me good today, was printing a case and the part of the wall that the slicer thought was bridge was not being printed before the bridge was started so the bridge just fell off into air. Turning off "Detect overhang walls" fixed it.

I dont see that as a real fix?  The point of detecting them is so the slicer can slow the print down, and increase the cooling, for those areas that need it.  There seems to be 2 actual problems here (maybe separate bugs, not sure)... 
1. Orca is simply not always correctly detecting overhang walls and placing them in odd spot. 
2. When it does properly detect them its not using bridge settings and they fail to print properly. Evidenced by printing success with much long/harder bridges using the same filaments.

CaptainFalcon63, 2024-02-02:
I ran into this today when using some built-in supports on a part I designed. The workaround I used was to place a modifier at the bridge location and change the settings to have 0 walls. As expected it wouldn't create an overhang wall if walls weren't being generated.

This is the first time I've experienced this but I looked at it as more of a "bridges not reaching walls" problem rather than a "bridges sliced as walls" problem. In the future it would be great to not need to create modifiers to fix the problem because built-in support can save time and material over slicer generated support if done correctly.

Jaze20, 2024-02-22:
1.8.0-RC2 same thing, project file included

Detect overhang walls turned off.  not a fix as noted above because of the speed.
[image]

Detect overhang walls turned on.  
[image]
[Heatbreak Exhaust Duct - Orca.zip](https://github.com/SoftFever/OrcaSlicer/files/14367180/Heatbreak.Exhaust.Duct.-.Orca.zip)

Make Overhang Printable  creates a solid block

danno131313, 2024-03-04:
I've noticed enabling "Classic mode" in the dynamic overhang speed settings still slices bridges as overhang walls, but it causes them to use the bridge speed setting rather than the overhang speed setting for 75%-100%:

[image]
[image]

obertini78, 2024-03-20:
Same issue here - v2.0.0-beta linux.
Bridge is being marked a bridge but speeds are set to those of overhang I guess. In the case below, speed goes down to 2 mm/s!
[image]

If I check the 'Classic mode' under 'Overhang speed', I get a 30-50 mm/s as expected.

kpipes556, 2024-05-26:
Same problem here. Trying to do a temp tower and the overhangs were too slow because they were using bridging speeds.

Turned off "Detect overhang walls." This fixed the overhang issue but created a bridging issue: outer and inner wall speeds were applied to the bridge.

[image]

Classic mode or not, I couldn't find a combination of settings to fix this and allow changing overhang and bridge speeds independently.

SoftFever, 2024-05-27:
> Same problem here. Trying to do a temp tower and the overhangs were too slow because they were using bridging speeds.
> 
> Turned off "Detect overhang walls." This fixed the overhang issue but created a bridging issue: outer and inner wall speeds were applied to the bridge.
> 
> <img alt="Screenshot 2024-05-25 at 9 52 52 PM" width="1377" src="https://private-user-images.githubusercontent.com/136215492/333842565-ecbf5a4f-4aa7-4447-b473-42f4f32d5c58.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4MjY4NzAsIm5iZiI6MTcxNjgyNjU3MCwicGF0aCI6Ii8xMzYyMTU0OTIvMzMzODQyNTY1LWVjYmY1YTRmLTRhYTctNDQ0Ny1iNDczLTQyZjRmMzJkNWM1OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyN1QxNjE2MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMzViOGJkOTNmNjYwZThhOTRkYzNjNTg0ZjlhN2Q3NGZkODgwMmVhYmZkNGRiNmIzYTA2Y2I0N2YzOTk0ZWRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Wag5Gifl83_Vwan5ULcNfzxrdB4eLoPASQ5r2VlGHHA">
> Classic mode or not, I couldn't find a combination of settings to fix this and allow changing overhang and bridge speeds independently.

This is most likely min layer time slowdown.
Try to set the "slow_down_layer_time" to 0 and try again

kpipes556, 2024-05-27:
> This is most likely min layer time slowdown. Try to set the "slow_down_layer_time" to 0 and try again

I tried turning off the min layer slowdown time and turned up the max volumetric just to take them out of the equation and it is still doing the same thing.

[image]

No matter what setting I try, it either adds walls to the bridge or doesn't let me set the overhangs independently and sets them equal to the bridge speed. For TPU, this just isn't working :(

2.0.0 on M1 Mac

mfish38, 2024-06-16:
Same issue on 2.1.0-beta linux. I have noticed that quite often orca slicer will try and print into thin air, sometimes even trying to make the first layer of a hole using a concentric pattern starting from the center of the hole working out towards the hole edge, which is just extrude a bunch of plastic into thin air and do nothing.

[image]

Felixoid, 2024-07-12:
I have the same issue in 2.1.1-1

[image]

It prints a completely hanging bridge, and I can't get rid of it.

[Buse_V6_BLT_ED3V6.tar.gz](https://github.com/user-attachments/files/16198376/Buse_V6_BLT_ED3V6.tar.gz)

vadixidav, 2024-08-10:
This issue causes issues with bridging. Notably, my bridge speed is very fast, at 150mm/s, but regardless of whether "detect overhang wall" is checked or not, these "walls" are printed at 25mm/s, which is too slow for bridging in this case, as I am printing very hot at this time. It is important that all bridging occurs at the same speed. Even if this is part of a wall, it is still a bridge. It is a bridge wall, not an overhang wall.

Ltek, 2024-08-24:
@SoftFever this bug has been here for a year or so. Would really appreciate if you can address this in Orca and somehow make this 'stupid proof' (for people like me).
If you believe there is a specific combo of setting that will avoid this problem, please have them set automatically (when changing a given setting/variable) so this doesnt happen **_or_** have a pop-up stating something like "you are changing setting ABC, in order to avoid slicing problem with Bridges you must set XYZ1 to blah and XYZ2 to bla-blah"

ShaneDelmore, 2024-09-22:
I ran into this today as well.  I am currently working around the issue by unchecking "detect overhang walls".  I ran into at least 2 issues with the feature, first, it slices my bridges as if they are overhangs, but 2nd, the tooltip says "for 100% overhangs bridging speed is used" but it actually used the overhang speed for 75-100% overhang.

vadixidav, 2024-09-23:
> I ran into this today as well.  I am currently working around the issue by unchecking "detect overhang walls".  I ran into at least 2 issues with the feature, first, it slices my bridges as if they [...]

This bug has been present for quite a long time now. I recommend making the largest overhang setting (75% to 100%) the same as your bridging speed and it fixes the issue. You will no longer be able to print anything going beyond 50% overhang (due to interpolation), but it improves print quality dramatically. I would just suggest not printing any parts that go beyond a 50% overhang, otherwise you have to fiddle with settings to work around the various bugs related to bridging speed.

hotellonely, 2024-10-11:
This bug is still presenting and quite sad. Someone has submited a fix for bambu studio and hopefully it can be integrated here.

https://github.com/bambulab/BambuStudio/commit/0efa2db823fa3b69cfde2d40b47b0724211f18d6

joek817, 2024-11-25:
ran across this too with a print:
[image]

but disabling "extra perimeters on overhangs" made it more logical:
[image]

not sure if that applies to your situations but hopefully it does

northcoastam, 2024-12-05:
Printing a temp tower I remembered this behaviour.  I found unchecking _Slow down for curled perimeters_  at least allows the bridges to print at bridge speed. I'm using v2.2.0
[image]

staal54a, 2024-12-09:
Bumping as I have hit the same issue in 2.2.0. I can work around it but actually being able to use bridge settings for bridges without giving up things like slowing down for curled perimeters would be preferred.

As a note, https://github.com/SoftFever/OrcaSlicer/issues/6448 appears to be a somewhat related feature request.

Rimjakk, 2024-12-16:
12/16/2024 -- Still happening. Showing overhangs where it should be bridging. 
[image]

Ltek, 2024-12-17:
@SoftFever this is a very pervasive bug, I suspect nearly all Orca users are affected by this.

hotellonely, 2024-12-17:
> @SoftFever this is a very pervasive bug, I suspect nearly all Orca users are affected by this.

It has been there for a very long time. In fact this has been an issue with Bambu Slicer as well but they half fixed it eventually (they gave this kind of walls the same print speed as bridges, but it still shows up as an overhang wall in UI). 

A dev in Discord promised to fix it but it hasn't come.

hungrierwtf, 2024-12-30:
Adding my example, from 2.2.0:
[image]
[image]
The section that should be a bridge gets sliced as an overhang wall, and the slicer gives the "floating cantilever" warning when slicing and uploading the print. It prints at normal flow and sags rather than stretching across the gap.

Rimjakk, 2024-12-30:
TBH, at this point I just turned off “slow down for overhangs.”  Whatever that Prusa proof test was which compared Bambu to whatever new hotness Prusa was pushing - where in the video it showed Bambu having super globby overhangs when slow down was turned off - apparently was completely false news.  I turn off slow overhangs and let the Bambu just shred as it pleases and, unsurprisingly, the accuracy of the thing wins out in the end and there is now of the fabled janky blobs Prusa had reported in their comparison video.  

So yeah. Turned off slow down for overhangs, and yeah, if I ever do find myself in the situation where I’m dealing with some gnarly cliff-hanger overhangs, I assure you I wont be depending on slowing down to do the job, Ill just use supports. 

It’s just dumb that bridges aren't being recognized as bridges in BS/Orca, but Cura (which is substantially less user friendly) doesn't seem to have a problem with this. 
This single issue is probably one of the top insidious causes for catastrophic failures, impacts and machine damage.  I know it is the *singular* reason for every major failure and impact Ive ever experienced in my tenure with Bambu via BS/Orca. … no, thats not true. BS/Orca cant figure out how to do bottom on-model interfaces on tree/auto either - but that’s a whole different topic..

vadixidav, 2024-12-30:
> TBH, at this point I just turned off “slow down for overhangs.”

Make sure you also uncheck "slow down for curled perimeters". This seems to be the cause of many print failures involving overhangs. It won't fix the other issues, but it solves this issue in most cases. Unfortunately, this means you will need to go very slow (like 20mm/s) on your outer wall to achieve consistent quality. Like you mentioned though, it is best not to rely on the slicer to do the right thing.

Rimjakk, 2024-12-30:
I have mixed emotions on that one and it has more to do with the printer being used. On the Bambu I think I turn it off because it runs hot enough to turn those edges flat or down leading up to our false-bridges. 
On the Ender printer, when I turned it off, that single feature resulted in a catastrophic failure due to the exact symptom named: curled perimeters. It wasnt the outside perimeters which curled, it was the “perimeter” leading up to the false-bridge which curled up more and more, I think due to the slower, cooler print temperatures of the Ender nozzle (we’re talking Ender 3 Pro, not one of the fancier ones). 
I haven't quite nailed down that issue yet other than deciding that it would probably be best to print those high speed geometrical and complex with the goodness that is the BambuLab, and leave the Enders for my low and slow open air jobs (such as printing dice towers or pretty things). 
The Bambu is a fine machine, to be sure, but for some reason the Ender printers just print glass surfaces with the 0.4 where the Bambu leaves the finest layer shift lines for whatever reason.  Nothing to complain about, just comparing identical prints. The Ender takes 10 times longer, but the results can be breathtaking in cosmetic (silk) applications.

segfly, 2025-01-09:
**I found a potential solution in Orca Slicer 2.2.0 by setting "Bridge Counterbore Holes" to any value other than "None" (such as "Partially bridged").**

Like many here are experiencing, Orca uses the wrong line type for bridge lines resulting in terrible bridging, especially compared to Cura which is the slicer from which I am trying to replace. None of the proposed solutions in this thread so far worked well enough for my expectations:

1. I don't want to turn off overhang detection because I still want slow lines for steep wall angles.
2. @SoftFever proposed reducing the wall loops count, but that still uses the wrong line type at the bridge perimeter while compromising strength/finish for the rest of the model.

I'm not entirely clear on what the "Bridge Counterbore Holes" setting is supposed to do as the wiki page linked from the UI does not exist. Nevertheless, without this setting enabled Orca applies the overhang wall rule to the wall lines (and if the wall loops is greater than the total lines to print causes the entire bridge becomes overhang lines):

[image]

After setting the "Bridge Counterbore Holes" to "Partially bridged", Orca then treats all the bridge lines correctly as bridge lines and the bridge speed/flow for bridging is used (note that the bridge settings may be negated if the filament "slow printing down for better cooling" is enabled).

This is the expected behavior for bridges in my opinion:

[image]

For anyone looking for the "Bridge Counterbore Holes" setting, it's under Quality->Bridging:

[image]

**Caveats:**
This setting fixed the bridging issues for me on 2.2.0 while still working with all the other settings I wanted such as overhang detection, extra perimeters on overhangs, etc. However, there are are a bunch of open reports of bridging/overhangs not working as expected in many different circumstances - such as #5724 #6806 and #6941 (and more). I was unable to recreate the reported issue in #6941 on 2.2.0 so some of these issue may have been since fixed.

In my opinion and according to the scattered reports, the overhang/bridging logic is clearly buggy. If you do use the "Bridge Counterbore Holes" setting to improve overall bridging, be aware that you may encounter other problems elsewhere depending on other settings and/or the model (though I couldn't find any unintended consequences in my test cases.)

hungrierwtf, 2025-01-09:
As @segfly mentioned, turning on "Bridge counterbore holes" seems to fix the issue, at least in the previously failing model I had used - not just the bridges over long gaps but also a couple shorter ones that had been printed as overhang walls instead of bridges. I still got the "floating cantilever" warning, but that's no big deal if the bridges actually get processed properly and use the bridge flow and speed settings. For now I've set "Bridge counterbore holes" to "Partially bridged" as the default in my profile.

vadixidav, 2025-02-12:
I figured that printing a benchy with 100 walls would work, but apparently not. This happens regardless of whether "Bridge counterbore holes" is None or not:

[image]

This is easy to reproduce. Set the walls to 100 and try to slice the benchy. Even such a ubiquitous model still has the roof of the benchy sliced as an overhang rather than a bridge. This is unrelated to the speed issue, as these overhang walls are printed at bridge speed, but in midair:

[image]

hotellonely, 2025-02-12:
yeah it's very inconsistent when it comes to bridging decisions. Also benchy with 100 walls are hilarious. Maybe @igiannakas can share some lights on this one as he might be the most knowledgeable person with bridging stuff XD

igiannakas, 2025-02-12:
Unfortunately bridges are generated after the walls of the model are generated - which means with a lot of walls you don’t leave enough space for the bridge infill. Hence you see what you see.

It’s a design choice that has been carried over from prusa slicer - basically st some point we need to decide what’s more important priority wise. If the user sets X walls as needed, then the slicer attempts to generate those X walls. If there is no space then bridges are skipped if that makes sense?

In any case the walls are printed with bridging speeds but their pattern follows the wall pattern they are attached to. 

This is a very tricky problem to solve unless we “ignore” the number of walls and don’t generate them where bridging surfaces exist. But then that opens a different can of worms code wise.

igiannakas, 2025-02-12:
Forgot to say - the reason priority to walls is given is to ensure external surface consistency and part strength as the number of walls influences that.

If you want to avoid this issue, using a layer height modifier is the way to go, to reduce the number of walls where your bridging layer happens and allow for the bridge to be extruded.

igiannakas, 2025-02-12:
By the way the new feature currently being tested in my PR allows for significantly better bridging by introducing a second bridge layer above the first one.

It won’t solve the problem you’re seeing for small bridges but it will for bridges where bridge infill can be generated. 

This is a hard problem to crack because you can have 100% overhang walls in a model area just because the wall is genuinely overhanging the perimeter for a small segment - think a complex curved model with overhangs. You can also have an 100% overhang wall when doing a genuine bridge which is defined as a straight line between two points. 

The issue is how do you distinguish between those two types when applying speed settings. We could apply the bridge speed to 100% overhanging perimeters globally, but then this would cause minor overhanging walls to print faster than their neighbouring partial overhang. Or we can apply the overhang speeds to the walls causing the first case to look great but the bridge being sub optimally generated.

Hence we default to the second one as in general prioritising external overhang quality is better than prioritising 4-6 wall wide bridge appearance. 

We could have an option to flip that behaviour, that’s easy, but then the user would have to actively choose that - overhang speed or bridge speed for overhanging walls if that makes sense. I could give that a try and see if makes sense, but in general we try to avoid adding options unless the benefit is meaningful and for small bridges I am not convinced it is.

igiannakas, 2025-02-12:
Let me take some time to give a more detailed breakdown:

**Issue 1: Speed handling of bridges:**
With curled perimeter detection, the overhanging perimeters are always assigned the overhang speed

[image]

Disabling curled perimeter detection causes the overhanging perimeters to use the bridging speed

[image]

That is because in complex models like the below, curled overhang detection benefits the print quality by ensuring small external bridges are treated always as overhangs. This prevents unnecessary speed up and slow down for micro segments where the external wall is 100% overhanging:

[image]

vs 

[image]

**Conclusion:** Pick the option that best suits your model. Organic models with overhangs benefit from the curled perimeter detection being enabled. Straight line geometric models benefit from the curled perimeter detection being disabled

**Issue 2: Concentric walls generated instead of bridges:**
This happens when you have a bridged "hole" that is smaller than the combined extrusion width of your generated walls.

[image]

This can be solved by applying height range/object modifiers like so

[image]

This approach also works well when there is not enough infill left to anchor the bridge if the model walls are thin.

**Conclusion:** If the user desires both large number of walls and bridging being generated, this can be achieved through the use of modifier to override the default priority of operation.

So in conclusion, I think that both issues mentioned in this GH issue are addressable through the use of the available options

dzid26, 2025-04-01:
> With curled perimeter detection, the overhanging perimeters are always assigned the overhang speed

But this is not right - it should not touch straight lines per the function name. 

And even if you turn off curled perimeter slowdown, it still displays them as overhangs, instead of bridges. 
[image]
(Albeit, it applies the bridge speed correctly indeed).  

Additionally, why turning on `Bridge counterbore holes` changes straight lines to bridges? 
[image]
This again should to a straight line. 

Seems like there might be an issue with "straightness" detection...

--------
Bonus bug I found: https://github.com/SoftFever/OrcaSlicer/issues/9183 - but it doesn't seem related.

Nishandh, 2025-04-01:
The problem I described here , related to internal_bridge : 
https://github.com/SoftFever/OrcaSlicer/discussions/9124)
I guess it might be related to this discussion ?

vasilky3, 2025-06-05:
> This can be solved by applying height range/object modifiers like so

I am trying to use suggested variant. Can you clarify what exactly should I tweak?  

[image]

"Bridge Counterbore Holes" not working. 

[image]

johyphenel, 2025-06-05:
Another one — Orca generates zig-zag bridge with nothing to anchor the sides over a gap: 

[image]

[image]

I'm running into issues where the counterbore setting that is best for one bridge on the Voron Stealthburner is worse / impossible to print on another :(

igiannakas, 2025-06-06:
Voron parts have been designed with 3D printing in mind.

Counter bore hole bridging is there to address poor CAD design where a hole is placed in mid air. This can be prevented in CAD by creating counterbore bridges in CAD as Voron parts do for example.

For the use of this feature beyond this case, I’ll go back to designing for printability. Any bridge to anchor successfully needs to have 1 perimeter spacing from the layer below exposed. This can be done in CAD by either designing the surface where a bridge should anchor to be more than N+1 lines wide, where N is the suggested wall count by the designer (eg 1.6mm for parts that need 3 walls to print). Alternatively design bridges that are less then 2*N lines wide - so bridges thinner than say 2.4mm. That would convert all bridges to walls hence will not need any bridge infill anchoring as the perimeters should continue to print over as a bridge.

Anything else than this will create bridging issues in pretty much all slicers as they all prioritise wall printing first then bridges. 

Designing for 3D printing is important - addressing the above in the slicer will result in compromises in other areas - eg strength of the wall where a bridge is extruded as it would need for example to reduce wall count to get the bridge to anchor. 

Or it would need to create a double seam next to each bridge by splitting the bridge into basically multiple objects and printing it separately - weakening the part in that area (this is what the bridge counter bore hole workaround does).

So in my view I don’t necessarily see this as a bug to be fixed but rather it’s a tradeoff - wall quality and consistency take priority over bridges. But that’s just me :)

vasilky3, 2025-06-06:
> For the use of this feature beyond this case, I’ll go back to designing for printability. Any bridge to anchor successfully needs to have 1 perimeter spacing from the layer below exposed. This can [...]

@igiannakas , Is where any extra information on bringing in CAD? 
Hard to understand text with no examples. Thx in advance.

johyphenel, 2025-06-07:
I should probably separate my thoughts a bit:

**Default bridging behavior**: 

> For the use of this feature beyond this case, I’ll go back to designing for printability. Any bridge to anchor successfully needs to have 1 perimeter spacing from the layer below exposed. This can [...]

> Designing for 3D printing is important - addressing the above in the slicer will result in compromises in other areas - eg strength of the wall where a bridge is extruded as it would need for example to reduce wall count to get the bridge to anchor.

One of the slicers main jobs is deciding on compromises. The operator sets a bunch of goals and constraints, and the slicer has to make the object printable as best it can while doing it's best to meet all constraints. Let's talk about other areas where the slicer makes choices that don't follow constraints:

- if the part is too thin horizontally for X walls, the slicer doesn't force X walls just because it's in the settings 
- if the part is too thin vertically for X top/bottom layers, the slicer doesn't force X layers just because it's in the settings

I don't see why we argue for printing "guaranteed to fail" bridges (those that start over thin air) as the correct behavior? 

Going back to the model I linked in https://github.com/SoftFever/OrcaSlicer/issues/2231#issuecomment-2932096642 — the slicer correctly decides to print a full rectangle across the whole gap area for the bridging layer, but then decides to start it over thin air. 

> So in my view I don’t necessarily see this as a bug to be fixed but rather it’s a tradeoff - wall quality and consistency take priority over bridges. But that’s just me :)

Can you explain why this is the right choice? An object with nice looking walls but a failed bridge goes in the trash. 

On the other hand, if you reduce the number of walls for the bridge area to leave room for the bridge to attach, you won't even be able to see the difference because the bridge will cover the area with the missing wall. 

**Counterbore hole behavior**

I agree that this feature **shouldn't** be related to this bug, so perhaps we've uncovered a separate bug: why does changing counterbore hole settings affect straight bridges? For the example in https://github.com/SoftFever/OrcaSlicer/issues/2231#issuecomment-2946788999 — imagine this Voron part had a counterbore hole that needed fixing, I wouldn't expect the slicer to also change all the straight bridges (and in this case make them unprintable). 

The problem is that this unrelated feature seems to be the only ways to make some objects / bridges printable, but since you can't set it for only part of an object you have to take the changes it makes to all bridges or none.

igiannakas, 2025-06-07:
Because typically you can’t selectively reduce the walls in the bridge area but rather the whole island where the bridge is extruded from. That would lead to a lower wall count and possibly a seam at the bridge (which is what happens when the counterbore hole workaround is selected).

segfly, 2025-06-07:
> So in my view I don’t necessarily see this as a bug to be fixed but rather it’s a tradeoff - wall quality and consistency take priority over bridges. But that’s just me :)

While that may be true, I simply want the slicer to explicitly give me the choice. As someone said above, a part with nice walls but failed bridges is still a failed print.

And while modifying the model may sometimes be the right call, the slicer shouldn't assume that its way is always the right way. It doesn't know my design, and maybe I know what I am doing, and in special cases I'd rather have a lower wall count and possibly a seam than to put a chamfer in my design.

The workaround works just fine for me - but it seems odd that the only way for users to toggle this bridging behavior is via some unrelated feature toggle.

dzid26, 2025-06-08:
Reminder: the [initial issue](https://github.com/SoftFever/OrcaSlicer/issues/2231#issue-1912612182) was about possible misclassification of the lines anchored in two points as "overhang walls" instead of "bridge" and applying unexpected settings to those lines. 

------

Regarding printing possible bridges in the air - obviously the algo could be smarter. Examples:

- 2 Wall loops - bad:
[image]
- 2 Wall loops with "bridge counterbore holes" - not great:
[image]
- 1 Wall loops - ok:
[image]

We know the last one is best, except the big cube is missing a inner wall.  It should be detectable that adding a second walls on this bridge will remove ability to anchor the internal bridge.

------
In other times (e.g. bridge with a narrow bottom), we get a line completely in the air and there are two (or rather four) seams. But this model could be sliced as a single loop: 
[image]

vadixidav, 2025-06-18:
I will just add that as someone that prefers to print with 100% walls, aggressive arachne settings, and I build voids, infill, etc. into the CAD design, this is quite difficult to manage. I showed the Benchy previously, but at the very least a good compromise until the patches from prusa or bambu are merged would be to have a checkbox that turns wall bridges into bridges for bridge speed. See this Voron Cube:

[image]

If it was just smart enough to know at this point to print this with bridge speed instead of at a snails pace, it would make a huge difference. I think for people printing in PLA this is less of an issue, but with PETG the bridges have to be printed faster.

Some settings would really help if we could get some alternative methods for detecting bridges. I think there isn't a one-size-fits-all. I think a very basic heuristic would be a setting that says "If an extrusion leaves the side of the print by Xmm, it will be treated as a bridge for speed purposes" would be perfect. The user could set the threshold for how far away something needs to be "floating" off of anything before it gets considered a bridge and not an overhang. If something is printing in mid-air, we can be pretty certain that, at least for certain circumstances/designs, it is either unintended, or the designer was trying to put a bridge there. In any design I am working on, you either have overhangs or bridges. It is a simple heuristic that if a line is a certain distance floating away from anything under it that it can be treated as a bridge for speed purposes. This simple setting would help a whole lot, and I imagine should be easy enough to implement. I can't say exactly how this would interact with supports though.

johyphenel, 2025-06-28:
Another example of the benchy problem from https://github.com/SoftFever/OrcaSlicer/issues/2231#issuecomment-1754051708 — Stealthburner bridges get sliced in an impossible to print way if you have `thick external bridges` turned on: 

`thick external bridges: off`
[image]

`thick external bridges: on`
[image]

[5 more replies]

closed 2026-10-07 by ianalexis