OrcaSlicer/OrcaSlicer #2235: Substitute G-code

stenc55, 2023-09-26:
V1.7.0 introduced option to print with 2in1out hotends. They use 2 extruders.
Problem is, on any extruder change, Orca inserts line:
set_pressure_advance advance=0 
Line is inherited from Prusaslicer. Prusa said it is because pressure advance interferes with ramming. Using this line causes "Unable to infer active extruder steppe" error and printer stops.
It does not interfere with my printer, ramming works as it should.

According to Kevin from Klipper, "The Unable to infer active extruder stepper error occurs if a SET_PRESSURE_ADVANCE command is sent without an EXTRUDER parameter and Klipper can’t determine which stepper to apply the pressure settings to.

In Klipper, the pressure advance settings are applied to a stepper and not to a hotend. So, if you want to use pressure advance when “belted_extruder” is active it is necessary to issue a SET_PRESSURE_ADVANCE EXTRUDER=belted_extruder ... command."

Disable ramming  disables  line insertion, but ramming function (helping tip form) is lost. That leads to clogging (filament is stuck in the path to nozzle).

**Which printers will be beneficial to this feature**
Any with Klipper and multiple extruders->single hotend

**Describe the solution you'd like**
Similar as Prusa has, ability to replace line of G-code with custom one, e.g. "set_pressure_advance advance=0 " to ";set_pressure_advance advance=0 "
or
Disable line "set_pressure_advance advance=0" insertion every time extruders are changed

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

kylek29, 2023-09-26:
I'd second porting over that whole ability that Prusa has to find/replace code and create custom snippets. It's a pretty hand feature set.

closed 2024-01-02 by github-actions[bot]

Serj121, 2024-01-20:
it would be great if we added the ability to replace the G-code as in PrusaSlicer. This functionality is very lacking

mattiras, 2024-09-18:
Yes this would be very useful, now I had to take harder way and use Post processing, which makes sharing profiles difficult.

LukaszMoskala, 2025-10-28:
As a workaround, you can put in printer.cfg:

```
[gcode_macro SET_PRESSURE_ADVANCE]
rename_existing: _SET_PRESSURE_ADVANCE
description: Ignore SET_PRESSURE_ADVANCE without EXTRUDER
gcode:
  {% if 'EXTRUDER' in params %}
    _SET_PRESSURE_ADVANCE {rawparams}
  {% endif %}
```

Quick testing shows that it works in my case