OrcaSlicer/OrcaSlicer #2239: Most Bambu Lab profiles missing elephant's foot compensation

belak, 2023-09-26:
**Is your feature request related to a problem? Please describe.**
When I swapped to a 0.6mm nozzle a few weeks ago, I started noticing elephant's foot. Originally I thought it was something I did, but looking at the profile, the 0.4mm P1S profile has elephant's foot compensation set to 0.15mm, while the 0.6mm P1S profile has it set to 0.

Frustratingly, Bambu Studio is also missing elephant's foot compensation for those profiles, so maybe it should be updated there first?

**Which printers will be beneficial to this feature**
Any Bambu Lab printers currently missing the setting.

**Describe the solution you'd like**
Add elephant's foot compensation for Bambu Lab printers.

**Describe alternatives you've considered**
I can definitely set this up in a personal profile, but it seems like it would be useful for everyone.

**Additional context**
Add any other context or screenshots about the feature request here.

belak, 2023-12-26:
Bump - also related to https://github.com/bambulab/BambuStudio/issues/2621

belak, 2024-03-27:
They've resolved the Bambu Slicer issue, so maybe the profiles could be updated from there?

neo-jayfeather, 2024-06-30:
All current 0.6mm P1S profiles have 0.15 elephants foot compensation, however some print profiles are missing that, like the 0.16 high quality a1m profile has it set to 0, and the 0.16 high quality p1p doesn't have it set. In addition, the property in the json files is called "elefant_foot_compensation" and around 40 files are missing a value for that. I'm not sure if thats intentional, but something is probably not right due to the addition of more profiles or something else.

closed 2024-10-07 by github-actions[bot]