OrcaSlicer/OrcaSlicer #2244: Role-based wipe speeds are much faster than perimeter speeds

killerline10168, 2023-09-27:
### OrcaSlicer Version

Release 1.7.0

### OS version

Windows 11 22H2

### Printer

Ender 3 Max 

### How to reproduce

Enable role-based wipe speed

### Actual results

Even though role-based wipe speed is enabled, it is still wiping at almost 3.5 times faster than the perimeter speed, which I suspect is why I have horrible seams that look under-extruded (off-topic).

[image]

This is showing the perimeter speed, which is correct.

[image]

Here it shows the wipe and its speed (140 mm/s compared to 25 mm/s!)shows

### Expected results

If understood correctly, the meaning of role-based wipe speed is to continue the travel along the printed perimeter with the same speed as the perimeter was printed. This clearly shows otherwise.

### Project file & Debug log uploads

[Retraction test.zip](https://github.com/SoftFever/OrcaSlicer/files/12740461/Retraction.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Naninani, 2023-09-28:
I wonder if this is related to all of a sudden, me started getting terrible stringing with PETG, which I never had before.

killerline10168, 2023-09-28:
> I wonder if this is related to all of a sudden, me started getting terrible stringing with PETG, which I never had before.

I'm no expert, but I'm imagining that the nozzle is now moving too fast while wiping, which doesn't let the nozzle relieve its pressure, which is the whole point of a wipe (along with reducing the seam visibility). 

In my case I get horrible seams (see my [reddit post here](https://www.reddit.com/r/FixMyPrint/comments/15gmm5h/why_do_my_seams_look_like_this_at_occasionaly/)), which I believe could be caused by this bug (I still have this issue, although less since I lowered my PA value).

killerline10168, 2023-11-22:
@Noisyfox I found out what the issue here is:

When role-based wipe speed is enabled, it basically copies what is typed in Outer wall speed. But if the printer needs to slow down because of minimum layer time, overhangs, etc. the wipe speed will still be what is set in the Outer wall speed and decouple from the actual print speed. 

The current printing speed is 30 mm/s because of low layer time.
[image]

Wipe is performed, but at 98 mm/s because of the higher set Outer wall speed.
[image]

Sounds like wipe speed doesn't calculate the actual speed to be used, but rather uses the same variable as is defined when setting Outer wall speed, or am I completely wrong?

killerline10168, 2023-12-07:
Bumping so this doesn't close. Have any developers had the time to take a look at this?

closed 2024-03-14 by github-actions[bot]

staal54a, 2024-11-03:
Has this issue been resolved or should it be re-opened?

hotellonely, 2024-11-19:
It should be reopened. It's still a issue that is ruining some of the seam performance.