OrcaSlicer/OrcaSlicer #2252: Calculation of filament use for every type of printing part

Eldenroot, 2023-09-28:
**Is your feature request related to a problem? Please describe.**
PS displays detailed info for used filament

[image]

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Add these info (including first layer time) into Orca Slicer

[image]

**Describe alternatives you've considered**
---

**Additional context**
----

kylek29, 2023-09-28:
Agreed, that and the icon menu at the bottom of your PrusaSlicer screenshot are handy features of PS, would love to see those ported.

Eldenroot, 2023-12-17:
Also I like the icons - it is easy to display/hide what you need... really handy!

Noisyfox, 2023-12-19:
The used filament weight & length is on my todo list, though I personally find the icons very hard to use. I prefer the Orca's way so that I don't need to move my cursor to each of the icon to show the tooltip to find the one I need. I just can't remember which icon is which.

Eldenroot, 2023-12-19:
Nice, I really like the visualization of the used filament for every "type".

Also I like the "first layer time" info... 

I am looking forward!

About icons - ok, makes sense :)

PR #3246 (2023-12-24): Show used filament length and weight per line type
[image]

Fixes #2252
closed 2023-12-24 by SoftFever