OrcaSlicer/OrcaSlicer #2253: THUMBNAIL_BLOCK

za-x-az, 2023-09-28:
**Which printers will be beneficial to this feature**
all klipper printers

**Describe the solution you'd like**
When thew need for manual gcode edit arives the THUMBNAIL_BLOCK is rather badly located at the start.

**Describe alternatives you've considered**
Move the THUMBNAIL_BLOCK to the end of the file like cura.

**Additional context**
Add any other context or screenshots about the feature request here.

discip, 2023-09-29:
@za-x-az
Seems to be a duplicate of #2200.
If so, please consider closing this one. 😊

za-x-az, 2023-09-29:
> @za-x-az Seems to be a duplicate of #2200. If so, please consider closing this one. 😊

You might be right, i did go a good searvh but i looked for the use of Thumbnail 😁 and well found nothing 🤷‍♂️

closed 2023-09-29 by za-x-az