OrcaSlicer/OrcaSlicer #2255: Add option to allow only one program instance at the same time

kb-elmo, 2023-09-28:
**Is your feature request related to a problem? Please describe.**
Right now opening another STL file through double clicking opens another new instance of OrcaSlicer, which causes all current unsaved settings to disappear in the new window. 
This can also lead to problems in the configurations when they are saved in different states from different instances of the same program.

**Describe the solution you'd like**
Add an option to the preferences menu to allow only one program instance at the same time, similar to what SuperSlicer alread has.
Opening another STL file through double-click should open the file in the already running OrcaSlicer instance instead of launching another new window.

Option in the preferences menu of SuperSlicer:
[image]

closed 2024-01-27 by github-actions[bot]

kb-elmo, 2024-01-27:
Would be nice if this could still be considered.

Completely ignoring issues and then automatically closing them is not a cool move btw...

ashreid23, 2025-02-04:
There is a massive call for this. As I understand it, as Orca is a fork, it is still available in Bamboo Slicer but was removed from Orca. Is there a reason not to 'add it back in'  - I'm just being overly simplistic to make the point.

dturner, 2025-02-07:
It's been added.

Preferences -> Allow only one instance. 

Bosh. Sorted.

ashreid23, 2025-02-07:
> It's been added.
> 
> Preferences -> Allow only one instance.
> 
> Bosh. Sorted.

am i missing something..?

[image]

..bosh

jbkuma, 2025-03-01:
Yes, it looks like you might be using an older version. There are a number of settings missing from that screen.

 Current version is 2.2.0

[image]

m404, 2025-03-04:
I use Orcaslicer 2.3.0-beta2 both on Windows and Linux (Fedora), and while the setting to only open one instance of Orcaslicer works as expected on Windows, Fedora sadly crashes when opening an 3mf file from a browser (for example) and another instance of Orcaslicer is already running.

`Mar 04 10:05:22 fedora io.github.softfever.OrcaSlicer.desktop[2343603]: [2025-03-04 10:05:22.955212] [0x00007f6ba0da33c0] [trace]   Initializing StaticPrintConfigs
Mar 04 10:05:22 fedora io.github.softfever.OrcaSlicer.desktop[2343603]: dbus[2]: arguments to dbus_message_new_method_call() were incorrect, assertion "iface == NULL || _dbus_check_is_valid_interface (iface)" failed in file ../../dbus/dbus-message.c line 1378.
Mar 04 10:05:22 fedora io.github.softfever.OrcaSlicer.desktop[2343603]: This is normally a bug in some application using the D-Bus library.
Mar 04 10:05:22 fedora io.github.softfever.OrcaSlicer.desktop[2343603]:   D-Bus not built with -rdynamic so unable to print a backtrace`