OrcaSlicer/OrcaSlicer #2260: Filament Profile Upload for Permanence

addohm, 2023-09-29:
**Is your feature request related to a problem? Please describe.**
Problem?  No.  Annoyance, very much.  

**Which printers will be beneficial to this feature**
X1C esp with AMS

**Describe the solution you'd like**
I'd like to be able to upload established filament profiles to the printer so that I don't have to repeatedly specify filament  profile and calibration factors every time I open the slicer.  Realizing that these values can and will change, the profile should be modifiable or at the very least overwritable.

**Describe alternatives you've considered**
There are no alternatives offered

kylek29, 2023-09-29:
You'd probably be better off submitting this on the BambuStudio git repo as OrcaSlicer can only interact with the API's their printer (and firmware) provides. 

If I'm understanding this correctly, it is something BambuLab said they may work on. Their initial implementation of the filament calibration persistence system ties the Filament Profile ID to a system one (the one it inherits from) and then stores the K-Value (Flow Ratio is passed back into the filament profile and saved at the slicer). It's one a lot of non-standard profiles will say "Generic PLA" (etc.) at the printer.

addohm, 2023-09-29:
> You'd probably be better off submitting this on the BambuStudio git repo as OrcaSlicer can only interact with the API's their printer/cloud (and firmware) provides. They also have a feature request section on their support forum.
> 
> If I'm understanding this correctly, it is something BambuLab said they may work on. Their initial implementation of the filament calibration persistence system ties the Filament Profile ID to a system one (the one it inherits from) and then stores the K-Value (Flow Ratio is passed back into the filament profile and saved at the slicer). It's why a lot of non-standard profiles will say "Generic PLA" (etc.) at the printer.

I've done that, and hope either Orca or Bambu (or both) makes something happen.

closed 2024-01-05 by github-actions[bot]