OrcaSlicer/OrcaSlicer #2270: Not possible to reorder objects, and print in desired order first layer

blonestar, 2023-09-30:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10 Pro

### Printer

Voron Trident

### How to reproduce

You have multiple objects on the plate and want to add a printing surface, a cube 0.2mm height object, for dialing in the Z offset.
Now try to make it print first, before other objects.

[image]

What I do is switching print sequence to "By object" and then there is ability to reorder objects. I reorder them and switching back to "By layer". But often that does not have any effect on printing object order in the first layer.
[image]
[image]

### Actual results

[image]
[image]

### Expected results

I must be possible to reorder objects and slicer to respect the printing order, not just in "By object" printing mode but in "By layer" as well.
If there is some other way to do that please let me know. Unless 

### Project file & Debug log uploads

I don't think debug log is needed for this.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

whitegfx, 2023-10-23:
+1 Yes exactly writen @blonestar.

pedrolamas, 2023-11-26:
You saved me from opening a new Feature Request for this exact same issue!

FWIW, I'm using Orca Slicer v1.8.0 on Windows 11.

One thing I did notice is that if I change the print order to "per object", change the order of the objects, and then back to "by layer", it will respect the order I have just set!

So seems to me that "by layer" is actually disabling functionality that probably works just fine!

bani6809, 2023-12-11:
Control over print order by layer is incredibly important when printing multicolor face down. If white text prints before the black surrounding color, the edges of the text are blurred. If you luck out and somehow get it to put black surround before the white infill, it works fine.

It currently seems impossible to manually get the ordering. It's pure luck if orcaslicer decides to slice exterior or interior objects first.

pedrolamas, 2024-03-12:
Issue still valid...

eafq78, 2024-04-02:
yep still valid, no way to select the order to print first
Will be useful to add emboss text and use the first filament to do flush then emboss then change filament, in this way you only need to switch 1 time.
I try different way to do it.
always print first the flush (main part filament) then the main part and finally the text, have to change filament 2 times for a single layer

blonestar, 2024-07-02:
Just tested in v2.1.1 and it works it seems.
Can anyone confirm?
cc @eafq78 @pedrolamas @bani6809 @whitegfx

pedrolamas, 2024-07-03:
Yes, I just tested this myself and indeed I can now re-order the objects when print sequence is "by object"!

Front my perspective, this ticket is now complete!

bani6809, 2024-08-25:
still does not work in 2.1.1
no matter how i order the objects, orcaslicer wants to do the outside first, then the inside text. results are horrible.

bani6809, 2024-08-25:
> > still does not work in 2.1.1 no matter how i order the objects, orcaslicer wants to do the outside first, then the inside text. results are horrible.
> 
> It seems like you describing some unrelated issue or something.

From the issue title:
Not possible to reorder objects, **_and print in desired order first layer_**

How to reproduce:
You have multiple objects on the plate and want to add a printing surface, a cube 0.2mm height object, for dialing in the Z offset.
**_Now try to make it print first, before other objects_**

This is exactly what I am trying to do. So it's exactly the issue. Orcaslicer does not let you change the order of objects on first layer.

It is not unrelated. It is exactly the same issue.

blonestar, 2024-09-09:
Now whatever I do it doesn't work for me. Not even old tricks works like first to cut (copy and delete) objects, then add a primitive object and finally paste those copied.
Orca v2.1.1

[image]
[image]

blonestar, 2024-09-09:
What I did to quickly overcome the problem I assembled all objects together except the primitive box which is used to dial in first layer.
[image]
[image]

Please note in that case is not possible to cancel single object during the print.

bani6809, 2024-09-09:
it would be nice if orcaslicer just offered option to obey explicit order of objects.

chasepeeler, 2024-10-08:
I'm having a similar issue on 2.1.1. I have a little tower that I use when switching filaments because my printer likes to spit out a bit of a blob when it first starts printing. As such it's important that it prints on the tower before the main print. If I just have the main model and 1 tower it seems to respect the print order without issue. If I add a second tower (because of two color changes) it totally ignores the specified order. Changes the placement of the objects doesn't make any difference either.

It won't let me attach a 3mf file, but I'll gladly provide it by other means if requested.

blonestar, 2024-10-08:
I'm experiencing whenever is more then 2 objects on the plate ordering was completely ignored.
Workaround for me sometimes is to assemble objects excluding the test print and then reorder, however that way assembled objects loose custom seem drawing.
Orca v2.1.1 and previous versions

ps We need to open the new ticket as this one is obviously ignored.

blonestar, 2025-01-08:
The issue is still present in the 2.2.0

blonestar, 2025-01-08:
Using Orca for a while now and didn't know about the option.
[image]

closed 2025-01-08 by blonestar

impeeza, 2025-10-16:
I can swear I did changed the order of print by rearranging the object order in the past. but on newer versions didn't work

by the way latest creality print indeed change the print order by the objects order.