OrcaSlicer/OrcaSlicer #2272: Infill direction changes Top Surface / Skin extrusion direction

Ltek, 2023-09-30:
### OrcaSlicer Version

1.7

### OS version

Win11

### Additional system information

Intel i5 gen10

### Printer

Artillery Sidewinder

### How to reproduce

Change Advanced -> Infill Direction (see highlight in screenshot)

[image]

### Actual results

The top layer extrusion direction will change based on the value set for Infill Direction. 

### Expected results

Top layer extrusion direction should not change based on the value set for Infill Direction. I'm also filing a Feature Request to 'Force Top Layer Skin extrusion direction" which is useful for designs that want/use the extrusion lines in the final design look.

### Project file & Debug log uploads

not necessary, happens with every/any project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ltek, 2024-01-01:
hello

Ltek, 2024-04-03:
still a problem

Ltek, 2024-07-04:
not stale... @SoftFever any chance you could add this? Cura has had this for years

Ltek, 2024-10-04:
still need this. does any dev even look at these FRs? been over a year and still not a mention of this from a dev... wow

SoftFever, 2024-10-07:
Can you post some use cases of such settings with preferable photos?

Ltek, 2024-10-08:
> Can you post some use cases of such settings with preferable photos?

the issue is that we cannot control the Top Surface / Skin direction so this setting, which is under Strength -> Advanced -> Infill Direction influences the top layer's direction. A 'strength' feature should be about strength, as most would assume. 

But if anyone wants to control the top surface directly for cosmetic reasons, there is no way to do that.  If you have a multi-level top-surface and you want all the layer lines to be horizontal, that is somewhat impossible. Or only will happen is you are 'lucky'

While this is not the best example it does show the problem., there is no possible way it seems (I've spend hours testing different settings, posting on Reddit, etc - no one knows how to do it) to get all the top surface extrusions the same direction. And when you change this Infill Direction, it changes the top layer's direction too, not just the "infill" (internal sparse or solid)
[image]

SoftFever, 2024-10-08:
> > Can you post some use cases of such settings with preferable photos?
> 
> the issue is that we cannot control the Top Surface / Skin direction so this setting, which is under Strength -> Advanced -> Infill Direction influences the top layer's direction. A 'strength' feature should be about strength, as most would assume.
> 
> But if anyone wants to control the top surface directly for cosmetic reasons, there is no way to do that. If you have a multi-level top-surface and you want all the layer lines to be horizontal, that is somewhat impossible. Or only will happen is you are 'lucky'
> 
> While this is not the best example it does show the problem., there is no possible way it seems (I've spend hours testing different settings, posting on Reddit, etc - no one knows how to do it) to get all the top surface extrusions the same direction. And when you change this Infill Direction, it changes the top layer's direction too, not just the "infill" (internal sparse or solid) <img alt="image" width="704" src="https://private-user-images.githubusercontent.com/1626802/374391467-bc171743-0cdc-4530-b45e-d6bba0367f72.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0MjgzMzgsIm5iZiI6MTcyODQyODAzOCwicGF0aCI6Ii8xNjI2ODAyLzM3NDM5MTQ2Ny1iYzE3MTc0My0wY2RjLTQ1MzAtYjQ1ZS1kNmJiYTAzNjdmNzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDhUMjI1MzU4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Nzg2NDEwMTFkODBhYmJlNTExMTJjOGIyNjQwNjE0YjcyNGI0NjFlYjI3ZmQyYjRlNjY0OTg0Y2E2YzZmOTA4ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.rAmJsOIJqf9FR1ZkHuRyVFCiO0BraLN1Uz6ZCilWe6c">

Let me rephrase my question: in what situations do you think the top surface infill direction matters, and why? Do you have any photos of real prints to demonstrate the difference?

Ltek, 2024-10-09:
> Let me rephrase my question: in what situations do you think the top surface infill direction matters, and why? Do you have any photos of real prints to demonstrate the difference?

Some large flat surfaces show the extrusion lines sot its ascetics. Hard to capture photos. Even with Matte black PLA which I use often, if you change directions of the lines it no longer looks as good... and can be very noticeable

Cura has had top and bottom layer direction for years...

[image]

closed 2025-04-15 by github-actions[bot]