OrcaSlicer/OrcaSlicer #2273: Force Top & Bottom Layer Skin Extrusion Direction

Ltek, 2023-09-30:
Would like to be able to force the top/bottom (visible layers) skin extrusion directions. This is very useful for designs that want/use the extrusion lines in the final design look. 

**Describe alternatives you've considered**
Alternative is to rotate the print on the bed, which is frequently not always possible for some size/shape items. Even then this doesn't guarantee both top and bottom will be the same direction

Ltek, 2023-12-30:
bueler

closed 2024-04-30 by github-actions[bot]

Ltek, 2024-04-30:
anyone home

mtc1122, 2024-04-30:
I think you may be referring to "Infill direction" which is under Strength>Advanced. This doesn't just refer to the sparse infill, but also refers to the solid infill on the top and bottom layers. You can change the angle of the lines/pattern on the top and bottom layers without having to rotate the model.