OrcaSlicer/OrcaSlicer #2278: Why are User Presets forcibly tied to System Presets?

Kaelum, 2023-09-30:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Additional system information

N/A

### Printer

N/A

### How to reproduce

You can't create any printer unless you base it on some other printer type.

1. Add a printer based on any System Preset, it doesn't matter what you chose.
2. Remove the System Preset.
3. The printer you created is gone.

### Actual results

PrusaSlicer allows you to create printers from nothing, and it also allows you to disconnect derivative printers from a System Preset.  For some reason this is no working, which makes Orca Slicer unusable.

### Expected results

Removing a System Preset should have absolutely no effect on User Presets.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

killerline10168, 2023-10-02:
1+

Ocraftyone, 2023-10-05:
Yes, PrusaSlicer allows you to "Detach from System Preset" which would probably be a nice addition. In the mean time, you can save your config (File>Export>Export Configs, then select a folder to save them to), then open the config file for your printer, and delete the key "inherits". You should then be able to save the config file, then import it into Orca (File>Import>Import Configs, select modified printer config file)

Kaelum, 2023-10-05:
@Ocraftyone I had already tried that, which is why I created this issue.  If you remove the "inherits" key, OrcaSlicer will respond with the following error:

`There are 0 configs imported. (Only non-system and compatible configs)`

And it will not work.

PhilippMolitor, 2023-11-02:
+1 for this feature, takes away a lot of pain when using OrcaSlicer with lots of different printers. It is a mess to always suffix all the other config parts with "@ Printer name".

kylek29, 2023-11-02:
+1 

I came across this a while back. I think it's likely a carry-over from the BambuStudio side of things (for syncing / their needs).

cragis, 2024-01-17:
+1

Would make my life easier :)

DiscusZ, 2024-01-29:
I see this is still an issue.    Why cant you add custom filament settings  and let any printer use them.  I have my ender 5 plus , it only allows me to access generic ABS   nothing else  and can not create ANYTHING else for it.

closed 2024-05-06 by github-actions[bot]