OrcaSlicer/OrcaSlicer #2280: Time Estimate Reaching Infinity When Added Filament Change on Layer

sabaatworld, 2023-09-30:
### OrcaSlicer Version

1.7.0

### OS version

macOS 13.6

### Additional system information

Machine: MacBook Pro 16 inch 2023 (Apple M2 Pro)

### Printer

Bambu Lab P1S

### How to reproduce

1. Add primitive Orca Cube.
2. Slice file.
3. Drag to layer 3 and add filament change.
4. Slice again.

### Actual results

Note the almost infinite time. If this GCODE is sent to printer, it crashes the printer on filament change and causes extruder to clog.

### Expected results

Time estimate should be correct. Print should be successful.

### Project file & Debug log uploads

[OrcaCube.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12775687/OrcaCube.3mf.zip)
[OrcaCube_PLA_35583992d72h49m.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/12775688/OrcaCube_PLA_35583992d72h49m.gcode.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

UsSuggester, 2023-10-04:
I have the same trouble.

In my case, when I use this option to filament change, my printer does the M600 (retraction) before starting the print, even before doing the skirt. I did check the starting G-Code again and its nothing like that on it.

At least "Add Custom G-code" works fine.

shlagwuk, 2023-10-14:
same bug: https://github.com/SoftFever/OrcaSlicer/issues/689

closed 2024-01-21 by github-actions[bot]