OrcaSlicer/OrcaSlicer #2283: 1.8-Dev - UI glitches with MacOS Sonoma

igiannakas, 2023-10-01:
### OrcaSlicer Version

1.8 Dev

### OS version

Mac OS Sonoma

### Printer

bambu X1 carbon

### How to reproduce

Have noticed the below GUI glitches when compiling from source on the latest code branch using Xcode 15 and MacOS Sonoma.

Issue doesnt exist on the pre-build binaries.

Any pointers as to what may be broken when compiling with Xcode 15?

[image]

[image]

### Actual results

As above

### Expected results

As above 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-10-01:
Also noticing that the Calibration tab for bambu printers is crashing again on Sonoma. Looks like something is weird with the compilation toolkit for Sonoma.

igiannakas, 2023-10-01:
Crash log for when clicking on the calibration tab: 
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OrcaSlicer [93024]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               1.8.0-dev (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-10-01 17:13:06.6081 +0100
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Anonymous UUID:        3F6AA9F3-CA5A-CD3B-8787-7AB612565EBC

Sleep/Wake UUID:       36FEC911-10B7-461A-943B-43A3479AC110

Time Awake Since Boot: 28000 seconds
Time Since Wake:       575 seconds

System Integrity Protection: enabled

Crashed Thread:        0  orcaslicer_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [93024]

VM Region Info: 0 is not in any region.  Bytes before following region: 4335353856
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102684000-10604c000    [ 57.8M] r-x/r-x SM=COW  ...OS/OrcaSlicer

Thread 0 Crashed:: orcaslicer_main Dispatch queue: com.apple.main-thread
0   libc++abi.dylib               	       0x183b13178 __dynamic_cast + 48
1   OrcaSlicer                    	       0x1031bdcc0 Slic3r::GUI::PressureAdvanceWizard::on_device_connected(Slic3r::MachineObject*) + 140
2   OrcaSlicer                    	       0x1031b351c Slic3r::GUI::CalibrationPanel::update_all() + 204
3   OrcaSlicer                    	       0x1052bdab4 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 224
4   OrcaSlicer                    	       0x1052bd7b0 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
5   OrcaSlicer                    	       0x1052bd5d4 wxEvtHandler::ProcessEvent(wxEvent&) + 56
6   OrcaSlicer                    	       0x1052bd424 wxEvtHandler::ProcessPendingEvents() + 240
7   OrcaSlicer                    	       0x10522b36c wxAppConsoleBase::ProcessPendingEvents() + 112
8   OrcaSlicer                    	       0x1052c5c48 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
9   CoreFoundation                	       0x183c370cc __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
10  CoreFoundation                	       0x183c36fb8 __CFRunLoopDoObservers + 532
11  CoreFoundation                	       0x183c365e8 __CFRunLoopRun + 776
12  CoreFoundation                	       0x183c35c2c CFRunLoopRunSpecific + 608
13  HIToolbox                     	       0x18e18e448 RunCurrentEventLoopInMode + 292
14  HIToolbox                     	       0x18e18e0d8 ReceiveNextEventCommon + 220
15  HIToolbox                     	       0x18e18dfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
16  AppKit                        	       0x187410f90 _DPSNextEvent + 660
17  AppKit                        	       0x187be4b94 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
18  AppKit                        	       0x18740443c -[NSApplication run] + 476
19  OrcaSlicer                    	       0x1051faad8 wxGUIEventLoop::OSXDoRun() + 112
20  OrcaSlicer                    	       0x1052c6318 wxCFEventLoop::DoRun() + 40
21  OrcaSlicer                    	       0x10524e698 wxEventLoopBase::Run() + 132
22  OrcaSlicer                    	       0x10522ac28 wxAppConsoleBase::MainLoop() + 120
23  OrcaSlicer                    	       0x1051c05e4 wxApp::OnRun() + 36
24  OrcaSlicer                    	       0x1052766e4 wxEntry(int&, wchar_t**) + 164
25  OrcaSlicer                    	       0x103377ef8 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 184
26  OrcaSlicer                    	       0x102696478 Slic3r::CLI::run(int, char**) + 26424
27  OrcaSlicer                    	       0x1026af444 main + 172
28  dyld                          	       0x1837e1058 start + 2224
[... 622 lines cut ...]
  ],
  "experiments" : [
    {
      "treatmentId" : "6dd670af-0633-45e4-ae5f-122ae4df02be",
      "experimentId" : "64406ba83deb637ac8a04419",
      "deploymentId" : 900000017
    }
  ]
}
}

Model: MacBookPro18,3, BootROM 10151.1.1, proc 8:6:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Aug  8 2023 04:27:44 version 20.10.1062.3.8.7.162 FWID 01-8e237e8f
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

Noisyfox, 2023-10-02:
Similar glitches happen on Windows as well when running directly from VS.

RaySajuuk, 2023-10-03:
Noticed this issue when compiled with Apple Clang 15.

SoftFever, 2023-10-06:
Yeah, I'm seeing similar weird stuffs after upgrading to Apple Clang 15...

igiannakas, 2023-10-06:
Are you also having the crash when selecting the Bambu introduced calibration page?

SoftFever, 2023-10-06:
Let me trigger the CICD for deps, which uses Xcode 14.

SoftFever, 2023-10-06:
> Are you also having the crash when selecting the Bambu introduced calibration page?

Hmmm, I haven't tried that.

igiannakas, 2023-10-06:
It doesn’t crash when the printer is offline, but crashes when the printer is online…

RaySajuuk, 2023-10-07:
> Yeah, I'm seeing similar weird stuffs after upgrading to Apple Clang 15...

Updating to newer dependency may resolve this issue. I suspect the issue is with wxWidgets.

closed 2024-01-13 by github-actions[bot]