OrcaSlicer/OrcaSlicer #2286: 0.2mm~0.8mm Profiles for Prusa MK3s+ and Mini

ucirello, 2023-10-01:
**Is your feature request related to a problem? Please describe.**
No a feature problem, but a project coordination problem. It seems that https://github.com/SoftFever/OrcaSlicer/issues/1333 slipped through the cracks, and the branch referred by @SoftFever is not accessible anymore (https://github.com/SoftFever/OrcaSlicer/tree/feature/consolidate_profile_naming)

**Which printers will be beneficial to this feature**
Prusa Printers

**Describe the solution you'd like**
I would like to see feature parity in terms of profile for Prusa Printers both in PrusaSlicer and OrcaSlicer

**Describe alternatives you've considered**
I have adapted a MK4 0.6mm profile to run on a MK3S+ Bear clone, and it worked OK. But in the past, I had better experiences when I copied from MK3S+.

przygos, 2023-10-21:
It would be nice to have those profiles

ucirello, 2024-01-20:
Ping?

IkarusX1978, 2024-01-24:
I'm also interested to see the Prusa profiles back in Orca Slicer!

SoftFever, 2024-01-24:
I will see if I can find some time to add them.

PR #4775 (2024-03-30): Various profile improvments 
Close #2286, close #3875, and partially #4761

Add Prusa MINI profiles (0.25,0.6,0.8), add MK3S+ nozzle types to generic filament types, & fix missing covers for Vzbot
closed 2024-03-30 by SoftFever