OrcaSlicer/OrcaSlicer #2289: Text and background have same color.

GAZ082, 2023-10-01:
### OrcaSlicer Version

1.7.0

### OS version

Linux - Gnome 44.5 Wayland

### Printer

X

### How to reproduce

Go to object settings:

[image]

### Actual results

Foreground and background have the same color.

### Expected results

Foreground white, background dark, as happens in the global menu:

[image]

### Project file & Debug log uploads

X

### Checklist of files to include

- [ ] Log file
- [ ] Project file

daduke, 2023-10-02:
same here, Fedora 38 wayland

TexZeTech, 2023-10-03:
Raised here https://github.com/SoftFever/OrcaSlicer/issues/2086

discip, 2023-10-04:
@GAZ082
Please consider closing this since it is a duplicate of #2086 as already mentioned by @TexZeTech. 😊

closed 2023-10-04 by GAZ082