OrcaSlicer/OrcaSlicer #2294: Is there any way to import settings from a gcode file?

simpat1zq, 2023-10-02:
Hello, this is a feature that I know SuperSlicer has from experience. I also believe PrusaSlicer has it based on a few forum posts I looked up. However, I don't think OrcaSlicer has it, or I just can't figure out how to do it. So if it's in there, could you please let me know how to use it. In Prusa, it's done by File > Import > Import Config and then pointing it to the gcode file. That same thing doesn't work on Orca. 

I even tried renaming the .gcode to .json(that's the only file type Orca accepts for config imports. 

But this is a super useful feature because there are times when I have a gcode file on my printer, and I want to just make a slight tweak to the settings that were used for it, but I would have to basically recreate them. Of course, I could save a project for each time I slice, but that is fairly cumbersome managing dozens of projects and linking them to the correct gcode file. 

Also, in the meantime, is there any way I could copy the settings from the gcode file(They are located at the end of the .gcode), and put them in a format that I can import into Orca?

WindArrow3d, 2023-12-31:
I have the same issue. Why on earth would anyone remove this feature? It has been standard on all OS slicers for many years....

liftbag, 2024-01-09:
It should be assigned as a feature request. I join the request, before github closes the topic due to inactivity.

simpat1zq, 2024-01-09:
Thanks. I didn't realize the feature was not there. I just assumed I didn't know how to use it since the feature is in super and prusa slicers.

digital0785, 2024-02-04:
this would be awesome if it's implimented i didn't realize it wasn't untill I went to go use it because my config got corrupt and was like... well balls lol

neilmendoza, 2024-02-16:
Just had an issue with my configs, thought could just import the gcode too as it works in Prusa but the feature is indeed missing :( Would be very useful to have.

jhenley01, 2024-03-10:
I requested this feature about a year ago. The status on it says done but it's still not working. I've also tried opening the gcode and then export config option and so far I haven't been able to get it working either.

https://github.com/SoftFever/OrcaSlicer/issues/575

simpat1zq, 2024-03-10:
> I requested this feature about a year ago. The status on it says done but it's still not working. I've also tried opening the gcode and then export config option and so far I haven't been able to get it working either.
> 
> #575

The status for that isn't done. It apparently got marked as an enhancement, and then it got auto-closed because there was no activity. 

@SoftFever it looks like you had marked that as an enhancement and added it to the dev plan. There seems to be quite a few people that want it. Is it possible to at least add it back to the radar?

WindArrow3d, 2024-06-09:
Bump. I still think this is an issue vs feature request. This functionality is available in all precursors to orca. I doubt my words here will make a significant impact, but I want it known that people are waiting for this to be fixed.

wallace11, 2024-08-05:
I second @WindArrow3d's comment.
This missing feature is a deal breaker.

Say you want to tweak the settings of some old gcode that you don't remember the exact settings you used for. Loading it into the slicer and then tweaking the required settings in PS and other derivatives is a breeze and a no-brainer. In Orca you'd need to dig into the end of the gcode in order to find the settings you used and hope that you didn't miss anything.

fborangic, 2024-08-13:
this would have saved my behind

JustSomeHuman, 2024-09-05:
I'd love to be able to import gcode directly into Orca Slicer.

jhenley01, 2024-09-11:
Just had an unexpected reboot and lost all of the settings to a profile I was working on. Sure would be nice if could just import them from the last print I ran instead of having to manually find them in the gcode file and set everything back up from scratch.

I'm really not sure why this keeps getting ignored by the developers, such a useful feature and I can't imagine it would be that difficult to implement since it's already in PrusaSlicer.

EuphoricPenguin, 2024-11-10:
Hear, hear. I hope this is something that is added.

simpat1zq, 2024-12-16:
And on top of all this, the config files, when exported aren't even in any sort of text format, so there's not even an easy way to parse the gcode and use that to create a config file for input into orcaslicer. Does anyone know of any sort of workaround? Like importing a gcode file into another slicer which can output in a format that orca can understand? I'm still surprised that there isn't a solution to this after this much time.

d0pexx, 2025-01-01:
I recently came across the same problem, no being able to import .gcode to Orca. But I was able to drag the .gcode file to the Orca-app and it worked without any issues.

digital0785, 2025-01-17:
this crap just hosed me again how is this still an issue .. it honestly makes me want to go to a different slicer.. I just had everything setup and dialed in for my coprint comp did an update over night and auto restarted started orca again and everything was wiped out.. don't even have my filament presets i created lol just the base profile that gets loaded from the printer..

Mart7Mcfl7, 2025-02-20:
Came here after trying to get settings from a previous slice. Absolutely ludicrous that this feature is unavailable, as it's been part of my previous slicers for many years. Honestly things like this, the obscure naming of features make me want to jump ship.

simpat1zq, 2025-05-22:
Not cool orca bot. Not cool at all.

WindArrow3d, 2025-05-22:
The only reason it's stale, is because it's being ignored... I'm still waiting for this feature to be FIXED.

simpat1zq, 2025-05-22:
So it looks like the feature is never going to be implemented. Does anyone know if there is a way to have orca save a version of the project everytime I upload to a printer?

WindArrow3d, 2025-05-22:
I'm going to try making a workaround. Not exactly sure how yet, but if I spend a few hours with chatGPT I might be able to figure it out.

simpat1zq, 2025-05-22:
You would be a hero to many if you did that. Even some kind of utility that converts the gcode settings into a file you can import would be amazing.

neilmendoza, 2025-05-22:
> I'm going to try making a workaround. Not exactly sure how yet, but if I spend a few hours with chatGPT I might be able to figure it out.

@WindArrow3d It's been a while since I used PrusaSlicer now but pretty sure it had this feature. As Orca is a fork of Bambu which is a fork of Prusa somewhere along the way the feature was removed. I don't think ChatGPT has a big enough context window but you could probably feed the whole codebase of each into Gemini and ask it to work out what needs doing to add the feature back in.

digital0785, 2025-05-22:
> [@WindArrow3d](https://github.com/WindArrow3d) It's been a while since I used PrusaSlicer now but pretty sure it had this feature. As Orca is a fork of Bambu which is a fork of Prusa somewhere along [...]

prusa and superslicer both do i believe which is kinda wild it would be left out this long LOL

simpat1zq, 2025-05-23:
> I think I have a useable workaround. It needs more testing. Once ready, I'll upload it to my github repositories and update you here.

That is awesome news. Please let me know if you need any help testing.

Mart7Mcfl7, 2025-06-17:
Would love to see this put back in

WindArrow3d, 2025-06-17:
Sorry everyone, I was mistaken when I said I had a useable workaround. It appeared to be working but I soon found out the issue is more complicated than I had thought. I haven't gotten the chance to come back to it but I do plan to eventually.

digital0785, 2025-06-17:
> Sorry everyone, I was mistaken when I said I had a useable workaround. It appeared to be working but I soon found out the issue is more complicated than I had thought. I haven't gotten the chance to come back to it but I do plan to eventually.

That sucks but I'm sure I speak for everyone when I say thank you for even looking into it to begin with more than I was able to do

simpat1zq, 2025-06-17:
> Sorry everyone, I was mistaken when I said I had a useable workaround. It appeared to be working but I soon found out the issue is more complicated than I had thought. I haven't gotten the chance to come back to it but I do plan to eventually.

No problem. You definitely got my hopes up, but I do understand that it might be more complicated.

Mart7Mcfl7, 2025-09-19:
> I implemented a workaround for this feature: My [GcodeTools](https://github.com/Matszwe02/GcodeTools) python library supports now extracting orcaslicer configuration files - so gcode spits out filament.json, machine.json, process.json (the same files that orcaslicer uses for its configuration) which is basically a few lines of python code:
> 
> import GcodeTools
> gcode = GcodeTools.Gcode('gcode.gcode')
> config = GcodeTools.Tools.generate_config_files(gcode)
> for key in config.keys():
>     with open(key, 'w') as f:
>         f.write(config[key])
> OrcaSlicer supports `--load-settings "machine.json;process.json" --load-filaments "filament.json"` but as for now it didn't work for me.
> 
> Update: new update works perfectly with loading these json files

Great stuff, I hope you get a lot of much deserved credit :)

digital0785, 2025-09-19:
> I implemented a workaround for this feature: My [GcodeTools](https://github.com/Matszwe02/GcodeTools) python library supports now extracting orcaslicer configuration files - so gcode spits out [...]

You sir are a gentleman and a scholar thank you for your work!

staffert, 2026-01-31:
Thank you!!
You saved my lost TEST profile. 
There seems to be some weird bug (or I did something wrong, more likely) in Orca Slicer 2.3.1.
It never seems to save this particular profile no matter what I do.....