OrcaSlicer/OrcaSlicer #2306: An M141 S0 appear at the end of the gcode no matter what your printer is (1.8.0-dev)

cpasoft, 2023-10-03:
### OrcaSlicer Version

1.8.0-dev

### OS version

Windows OS

### Printer

FLSUN V400

### How to reproduce

Just slice any model.

### Actual results

[image]

### Expected results

The M141 must not be there

### Project file & Debug log uploads

Not necessary

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-10-03:
It is probably a residue from Bambu extending their code to cater for the x1E that will have a heated chamber.

gbyleveldt, 2023-10-03:
Yeah, just noticed that as well this afternoon. Weird

cpasoft, 2023-10-04:
> It is probably a residue from Bambu extending their code to cater for the x1E that will have a heated chamber.

This makes sense.

discip, 2023-10-04:
Second that! 😊

PR #2328 (2023-10-05): Fix M141 S0 at end of gcode
fixes #2306 where gcode to turn off chamber heating is hardcoded into the slicer. BBS has also removed this line of code in their source.
closed 2023-10-05 by SoftFever