OrcaSlicer/OrcaSlicer #2309: Make internal bridge controls able to be independent of default bridge controls

Salmonsquire, 2023-10-03:
When printing large surfaces, I am having difficulty in dialing in values for the internal bridge layer that don't also impact the external bridges. The result is that there is poorer top surfaces, with occasional pillowing or overextrusion. The current method to resolve this is to lower the bridge width and bridge density. This reduction improves the first solid infill layer.

This would affect all printers

The solution I propose is to make it so that internal bridges are able to dialed in independently of default/external bridges. (Possibly even additional control for the interaction between internal bridge and the first solid infill layer)

discip, 2023-10-04:
Seems to be a duplicate of #2295.
If so, please consider closing this one. 😊

Salmonsquire, 2023-10-04:
> Seems to be a duplicate of #2295. If so, please consider closing this one. 😊

There is certainly overlap but the ticket #2295 is only asking for pattern to be independent where as I'm asking for flow, density, and possibly even solid infill/bridge overlap control.

Without changing the other ticket, I don't think it makes sense to close this one.

hippotastic, 2023-10-06:
I'd also love to see this feature. I just spent 30 minutes trying to figure out why my internal bridges were printed at 0.4mm layer height although the rest of my print is set to 0.25mm layer height and "thick bridges" are turned off.

Just like @Salmonsquire reported, the unexpectedly thick internal bridge caused a visible defect in my print as well - a big visible bulge in the inner surface of a bowl. The model was from the "Bauhaus tray" set on Printables.

PhilBaz, 2023-10-06:
Hey, this is basicaly a duplicate of my existing issue, and I have more detail. can you close this please a comment on my issue.

https://github.com/SoftFever/OrcaSlicer/issues/1110

Salmonsquire, 2023-10-06:
@PhilBaz @hippotastic Closing the issue and moving it to #1110

closed 2023-10-06 by Salmonsquire