OrcaSlicer/OrcaSlicer #2314: Nozzel heats up in mid air before print starts for no reason and waste of time

ddden9843, 2023-10-04:
### OrcaSlicer Version

1.7

### OS version

windows 10

### Additional system information

CPU: Ryzen 5 2600x
GPU: Rx590 Amd
Ram: 32gb ram

### Printer

Ender 3

### How to reproduce

any print (stl file) 

### Actual results

first it would auto home then start heating up to 150c and raise up to around 57mm(z-axis only) then it would heat up to the 205 I set it to, then it would auto home and after all that it would start the print same file from cura just heats the nozzle without moving 

### Expected results

Just heat up in the auto home position without doing all the unnecessary moving up heating then back down

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ocraftyone, 2023-10-04:
So what you have highlighted in the screenshot is part of the end sequence of the object when it moves the nozzle up and out of the way so you can easily get the part off the build plate. As for the start of the print, I have checked the start gcode and it heats up at 50 mm. If you want to change where the nozzle is when it heats up, you can change this in your machine gcode. As for the ending gcode, I would recommend not changing it unless you want to get into relative positioning.

To change start gcode:
Select edit next to the printer profile
[image]

Make sure advanced is ticked in the upper right corner of the printer settings dialog
[image]

Select "Machine G-code"
[image]

Under "Machine start G-code", change the z value on the line shown in the picture. It should be noted that it is a good idea to have the extruder off of the build plate during this process. I would suggest 10-20mm.
[image]

ddden9843, 2023-10-04:
Thank you so much that solved it

discip, 2023-10-04:
@ddden9843
As this is solved, please don't forget to close this issue. 😊