OrcaSlicer/OrcaSlicer #2316: Bed not centered in slicer

ddden9843, 2023-10-04:
### OrcaSlicer Version

1.7

### OS version

windows 10

### Additional system information

CPU: Ryzen 5 2600x
GPU: Rx590 Amd
Ram: 32gb ram

### Printer

Ender 3

### How to reproduce

I picked ender 3 and set the printable area 220x220 like default. excluded bed area 0x0. 

### Actual results

The bed is not centered and the prints, print way to close to the front edge(where the screen is, logo on bed etc.front)

### Expected results

Print in the middle and have the slicer show the bed is in the middle

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ocraftyone, 2023-10-04:
Are your models being arranged to close to the front of the bed or are you talking about the black square thing on the bed?

ddden9843, 2023-10-04:
both, the print is being printed really close to the front of the build plate and I have no idea what that black square is and why its off center I think they both might cause the same problem but I'm not sure

Ocraftyone, 2023-10-04:
I believe the black square is normal. On other printers, i believe it is replaced with a representation of the build plate (like an image). As for them being printed close to the front of the build plate, are you able to provide a screenshot? I didn't have that issue when I tested it.

ddden9843, 2023-10-05:
> I believe the black square is normal. On other printers, i believe it is replaced with a representation of the build plate (like an image). As for them being printed close to the front of the build plate, are you able to provide a screenshot? I didn't have that issue when I tested it.

Is there a way to get rid of the black square above my bed?

Ocraftyone, 2023-10-07:
@SoftFever may have a better idea. I'm not too well versed in that part of the slicer quite yet.

SoftFever, 2023-10-07:
It's probably caused by a bug in AMD graphic card driver.
Non square(or power of 2 size, I can't remember which one) textures can't render properly anymore since one of their driver updates.

Ocraftyone, 2023-10-07:
That makes sense. I run and as well.

ddden9843, 2024-01-11:
Is there a way to fix the rendering bug?

ddden9843, 2024-02-06:
> I too am experiencing this issue. Print preview shows everything centered on the bed, but send the print job to the printer it's positioned back and to the right. This is the same for all of my Klipper printers (Enderwire, v0, Trident, E3P x2, BiQu-B1). Currently on build 1.9

Still can't figure out why it's happening but I have a Ender 3 and its running Marlin but when I arrange the model in the middle in the software it does not print in the middle irl. as you can see in the picture there is a black square above the silver one I don't know how to get rid of it and for some reason the software thinks that's the build plate even tho its the one underneath

Kruworkz, 2024-02-13:
New to ORCA slicer, I'm having a similar issue. modified ender 5, BTT M8P running Klipper. the 1st 2 ORCA calibration prints, printed in the center as the slicer showed, now all of a sudden they are printer in the front left corner of the print bed.

closed 2024-05-21 by github-actions[bot]