OrcaSlicer/OrcaSlicer #2317: Support Error due to "Reload From Disk"

TheAtomicSoul, 2023-10-04:
### OrcaSlicer Version

1.7.0

### OS version

Win11

### Additional system information

na

### Printer

Bambu p1p

### How to reproduce

Load a model, do something that affects it height and reload it with "reload from disk"

### Actual results

The  Tree support "grows" into the model.

### Expected results

Tree support not printing inside the model

### Project file & Debug log uploads

na

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TheAtomicSoul, 2023-10-04:
Removing "Variable Layer Height" painted weights removed the problem.

Noisyfox, 2023-10-04:
This has been fixed in 7ef43f6849a878dead4cab9ed399aeac44b7cd24

TheAtomicSoul, 2023-10-04:
> This has been fixed in [7ef43f6](https://github.com/SoftFever/OrcaSlicer/commit/7ef43f6849a878dead4cab9ed399aeac44b7cd24)

Is that for a future update? (I'm not familiar enough with Github to tell.)

TheAtomicSoul, 2023-10-04:
sorry, still learning and didn't mean to reopen/.

discip, 2023-10-04:
> sorry, still learning and didn't mean to reopen/.

Tan please close again. 😂

discip, 2023-10-04:
@TheAtomicSoul
> sorry, still learning and didn't mean to reopen/.

Than please close again. 😂