OrcaSlicer/OrcaSlicer #2319: Request: printer profile for Creality Ender 3 Max Neo

chimericdream, 2023-10-04:
**Which printers will be beneficial to this feature**
Creality Ender 3 Max Neo

**Describe the solution you'd like**
Official profile for the printer

**Describe alternatives you've considered**
Using the generic profile and customizing it based on the Max Neo's specs.

**Additional context**
I am new enough to printing that I don't know how different the settings for the Ender 3 Max Neo are from the other Ender 3 variants. So while I know that I can technically use the generic profile, I am not yet fully comfortable doing so.

egelliott3, 2023-11-16:
Ditto.  I just downloaded Orca after seeing a video by The Next Layer, but no slicer for the Max Neo and Google was not helping either.

closed 2024-02-23 by github-actions[bot]

Duck-Lady, 2024-06-30:
Is there anything being developed for this?

dmcguire80, 2024-06-30:
This is the one I used as a base. For the 4.2.7 board some of the pins are inverted so test before pushing it. https://github.com/CrealityOfficial/Creality_Sonic_Pad/blob/main/printer_configrations/printer-Ender3maxNeo-103.cfg