OrcaSlicer/OrcaSlicer #2324: Non organic tree supports slice solid for ender 3 s1 profile

billy12, 2023-10-04:
### OrcaSlicer Version

1.7.0

### OS version

windows and osx

### Additional system information

mac osx big sur 11.3 intel i5
windows 11 intel i7

### Printer

Ender 3 s1

### How to reproduce

1. enable supports
2. choose tree supports 
3. use anything other than organic
4. try to print

### Actual results

[image]
when sliced the supports are completely solid and even over extrude when printing. this same issue happened on my ender 3 profile as well. but this issue doesn't happen on my sidewinder x1 profile. organic still works as it should which is good 

### Expected results

[image]
ive uploaded a picture with the same model same setting only difference is the printer profile, the expected result is a semi hollow support that prints normally.

### Project file & Debug log uploads

[low_poly_support hot mess.zip](https://github.com/SoftFever/OrcaSlicer/files/12807302/low_poly_support.hot.mess.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/12807314/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

GonzaloVillalobo, 2023-11-26:
I do have the same problem but when i change the "base pattern spacing" it turns to completely hollow and i cannot set any páttern or infil ratio or spacing on the support trees

closed 2024-03-04 by github-actions[bot]

crookz, 2024-03-04:
can this be reopened?