OrcaSlicer/OrcaSlicer #2325: Awful performance

uhthomas, 2023-10-04:
### OrcaSlicer Version

1.7.0

### OS version

Linux paradise 6.5.5-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 23 Sep 2023 22:54:46 +0000 x86_64 GNU/Linux

### Additional system information

```sh
❯ neofetch
                   -`                    thomas@paradise
                  .o+`                   ---------------
                 `ooo/                   OS: Arch Linux x86_64
                `+oooo:                  Host: X570 AORUS ELITE -CF
               `+oooooo:                 Kernel: 6.5.5-zen1-1-zen
               -+oooooo+:                Uptime: 1 hour, 16 mins
             `/:-:++oooo+:               Packages: 1020 (pacman)
            `/++++/+++++++:              Shell: zsh 5.9
           `/++++++++++++++:             Resolution: 3440x1440
          `/+++ooooooooooooo/`           DE: GNOME 44.5
         ./ooosssso++osssssso+`          WM: Mutter
        .oossssso-````/ossssss+`         WM Theme: Adwaita
       -osssssso.      :ssssssso.        Theme: Adwaita [GTK2/3]
      :osssssss/        osssso+++.       Icons: Adwaita [GTK2/3]
     /ossssssss/        +ssssooo/-       Terminal: tmux
   `/ossssso+/:-        -:/+osssso+-     CPU: AMD Ryzen 9 3950X (32) @ 3.500GHz
  `+sso+:-`                 `.-/+oso:    GPU: NVIDIA GeForce RTX 3080 Ti
 `++:.                           `-/+/   Memory: 12997MiB / 64225MiB
 .`                                 `/
```

### Printer

QIDI X-Max 3 (N/A though)

### How to reproduce

1. Use Orca Slicer on Arch Linux (https://aur.archlinux.org/packages/orca-slicer-bin)
2. Load some STL files (in this case; https://www.printables.com/model/78497-gourdo-the-pumpkin, https://www.printables.com/model/564262-portable-cable-organizer-winder).
3. Slice.

### Actual results

Try to use the UI at all - it will be completely frozen.

[Screencast from 2023-10-05 00-24-21.webm](https://github.com/SoftFever/OrcaSlicer/assets/9749173/428a2c74-5f9c-449b-880d-cd28ef44379a)

[Screencast from 2023-10-05 00-32-18.webm](https://github.com/SoftFever/OrcaSlicer/assets/9749173/281bce72-5478-40a5-b96d-e4ad443456e7)

### Expected results

It should perform reasonably well, as it does on macOS.

### Project file & Debug log uploads

[debug_Wed_Oct_04_23_52_08_9627.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12810861/debug_Wed_Oct_04_23_52_08_9627.log.zip)

[out.zip](https://github.com/SoftFever/OrcaSlicer/files/12810864/out.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-10-07:
Does it perform better if support is disabled?

w3z315, 2023-10-07:
Got the same issue. It's occasional, completely freezes the UI and I have to restart OrcaSlicer. Usually after starting a print and moving the OrcaSlicer to another display, then move it back to the original monitor and it completely freezes. I'm also using mixed monitor refresh rates (60hz / 165hz) 

**OS Info**
```
~ ❯ neofetch                                                                                                                                                                      
            .-/+oossssoo+/-.               xxxx@xxxx 
        `:+ssssssssssssssssss+:`           ------------------ 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 22.04.3 LTS x86_64 
    .ossssssssssssssssssdMMMNysssso.       Host: Z690 GAMING X DDR4 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 6.2.0-34-generic 
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 1 hour, 25 mins 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 2234 (dpkg), 17 (flatpak), 16 (snap) 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: zsh 5.8.1 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 2560x1440 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   DE: GNOME 42.9 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   WM: Mutter 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   WM Theme: Adwaita 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Theme: Yaru-blue [GTK2/3] 
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/    Icons: Yaru-blue [GTK2/3] 
  +sssssssssdmydMMMMMMMMddddyssssssss+     Terminal: gnome-terminal 
   /ssssssssssshdmNNNNmyNMMMMhssssss/      CPU: 12th Gen Intel i7-12700K (20) @ 4.800GHz 
    .ossssssssssssssssssdMMMNysssso.       GPU: AMD ATI Radeon RX 6800/6800 XT / 6900 XT 
      -+sssssssssssssssssyyyssss+-         Memory: 5478MiB / 31920MiB 
        `:+ssssssssssssssssss+:`
            .-/+oossssoo+/-.                                       
                                                                   
```

**Screenshot**
[image]

uhthomas, 2023-10-07:
> Does it perform better if support is disabled?

I presume you mean print supports? They don't maske a difference to performance.

ColdBeer72, 2024-01-09:
Trying to reopen the issue. The issue is still there. Ubuntu 22.04.3 (and AppImage version).

Options in the left column and 3D movements with the mouse are totally frozen. I could save the work but it's happening more frequently with every version, so it's totally annoying.

Please, review the problem or give us a way to make a debug or see some log so could help.

My suspicion is that there is some trouble with graphic library and Wayland (the graphic composer in Ubuntu).

RebeccaStevens, 2024-01-12:
Same issue for me on Nobara 39 (Fedora 39 based).

I would try building from source to see if that makes a difference, but I'm not familiar enough with such things to know how to do such a thing.

Skaifer, 2024-01-15:
Does this happen on X11 as well? I have a similar situation under Wayland. The app starts ok, but when I open project with plates and models, it barely responds to input. Very slow. Probably because I have Nvidia, and it behaves weirdly in apps under Wayland, and browsers require additional env variables for acceleration to work. Even then, UI elements in some apps, from time to time, disappear and reappear for no reason.

GlovePuppet, 2024-01-31:
Similar issue on Mint 20. Small, simple stl files will load and print but anything more complex than an XYZ cal cube hangs in the "Send print job to" dialog. Unfortunately this makes Orca unusable on Linux for me :(

RebeccaStevens, 2024-02-01:
Note: The 2.0.0-dev version is a lot better but not perfect.

vansoest, 2024-03-26:
Same issue here with xwayland on Hyprland nvidia-550 using 3 monitors. 
freezing but the app itself runs—I can close the window, the save dialog will pop up.

Any suggestions to compile the slicer for native wayland?

closed 2024-07-02 by github-actions[bot]

aclowe10, 2024-12-15:
Similar issues with Nobara 40 (based on Fedora 40) after latest Mesa updates. Weird as it was performing fine up till this update despite being on latest Nvidia drivers. Not entirely sure it's specifically mesa as Bambu Studio was working fine while OrcaSlicer wasn't but suddenly developed the same problems today.