OrcaSlicer/OrcaSlicer #2327: Tree support overlaps model with variable layer height

StanislavChumarin, 2023-10-05:
### OrcaSlicer Version

1.7.0

### OS version

macOS 13.6

### Printer

Creality Ender-3 S1 Pro 0.4 nozzle

### How to reproduce

1. Load curvy model
2. Enable variable layer height
3. Enable Support. Type: tree (auto)
4. Slice plate
5. Observe interaction of support and object

### Actual results

Support layers and object layers are overlapped or supports empty space (way below expected level)
[image]
[image]

### Expected results

Support structure must not overlap with object and support model

### Project file & Debug log uploads

[wrong_tree_support.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12816766/wrong_tree_support.3mf.zip)
[debug_Thu_Oct_05_11_37_49_32065.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/12816767/debug_Thu_Oct_05_11_37_49_32065.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Ocraftyone, 2023-10-05:
The latest 1.8 beta now displays an error stating that variable layer height is not compatible with organic supports.

cramsvik, 2023-10-12:
Loogs to me like layer height for the tree supports is static (at lowest? at slicer setting?) while some of they layers are way higher. This causes the supports to be way too short to reach where they are supposed to be, and it also caused the supports to merge into the model at some places, probably because the "scaled down" curve of the supports don't match the curve on the model.

[image]

[image]

[image]
[image]

arittner, 2023-10-21:
> Loogs to me like layer height for the tree supports is static (at lowest? at slicer setting?) while some of they layers are way higher. 

Yeah, saw also the same bug. I tried to add tree support for very tall components and failed with variable layer height:

[image]

The component in front is with variable layer height and adaptive calculation. The component behind with static layer heigth.

closed 2024-01-28 by github-actions[bot]

evilDave, 2024-08-29:
I have just struck this issue also.

Supports generated with variable layer height and independent support layer height both selected intersect the model. Some support parts also printed in space (even with independent support layer height off)

I noticed this with a fairly complex model. I have provided the 3mf file without the model added, the model that I was using is this one: https://cults3d.com/en/3d-model/art/triceratops-skull. Notice collisions at layer 565 (and just before) for example (there are others), and support material in free space on layer 651

Collision:
[image]

Odd support material:
[image]

Without independent support layer height:
[image]

[tree_support_issue.3mf.zip](https://github.com/user-attachments/files/16793775/tree_support_issue.3mf.zip)