OrcaSlicer/OrcaSlicer #2329: Prints in mid air - no "support needed" error showm

simonbuehler, 2023-10-05:
### OrcaSlicer Version

1.7

### OS version

WIn 10

### Printer

Bambu Lab P1P

### How to reproduce

[clutch-roller.zip](https://github.com/SoftFever/OrcaSlicer/files/12818876/clutch-roller.zip)
slice the model, the overhang would need support but it just says "slice ok", path generated:

[image]

### Actual results

print would fail obviously

### Expected results

warning: support needed

### Project file & Debug log uploads

[clutch-roller.zip](https://github.com/SoftFever/OrcaSlicer/files/12818923/clutch-roller.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kylek29, 2023-10-05:
Is this with Arachne? I had a very similar model that did the same thing, I checked it in another slicer and it was the same (if I remember correctly).

simonbuehler, 2023-10-10:
I just rechecked and it's both Arachne and classic

closed 2024-01-17 by github-actions[bot]