OrcaSlicer/OrcaSlicer #2330: Invalid Invite - Discord server

bankh, 2023-10-05:
### OrcaSlicer Version

1.7.0

### OS version

Ubuntu 20.04

### Printer

Ender3, Bambu P1P, ...

### How to reproduce

Click the Discord link on the repository.

### Actual results

Error received as follows:  
[image]
follows

### Expected results

Access to discord server.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-10-06:
Thank you.
I have updated the link

bankh, 2023-10-06:
It works. Thank you.

closed 2023-10-06 by bankh

itsmichaelk, 2023-11-15:
Need to re-open this - the link is expired again. Really would love to join the community.