OrcaSlicer/OrcaSlicer #2333: Orca slicer can't properly use python post proccesing scripts

Nathan22211, 2023-10-05:
### OrcaSlicer Version

1.7.0

### OS version

windows 11

### Printer

voron trident

### How to reproduce

1. put in a python post proccesing script like so `C:/post_scripts/arc/Scripts/python C:/post_scripts/prusa_slicer_post_processing_script`. the first part is to point to my pyvenv with the script's dependencies, like you would in prusa slicer. the 2nd part points to the script in question. change this as needed but note that for python based post proccesers to work, it needs to point to the python interpreter in the command
2. slice a file
3. attempt to export

### Actual results

it'll mention that `C:/post_scripts/arc/Scripts/python` is not a valid script. But that command would work normally in prusa slicer.

### Expected results

it should use the post script, arc overhangs in this case, as intended

### Project file & Debug log uploads

I don't see how a project file is helpful here. it pertains to the post processer so the paths will vary wildly depending on the system and user

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jmlott, 2023-10-06:
I use post processing scripts with Orca. My guess is that you didn't change the variable names from what Prusaslicer uses to what Orca uses. Have you looked at Klipper logs to see if that gives an indication as to what isn't working?

SoftFever, 2023-10-06:
Can you try to specify the python application?
Some thing like this:
`path_to_your_python/python.exe path_to_your_script/some_script.py`

Nathan22211, 2023-10-06:
got further, but unless I forgot to install the needed libraries I got this this error`Post-processing script C:/post_scripts/arc/Scripts/python.exe C:/post_scripts/prusa_slicer_post_processing_script.py on file C:\Users\carol\AppData\Local\Temp/orcaslicer_model/Thu_Oct_05/18_34_28#26020#40/Metadata\.26020.0.gcode.pp failed.
Error code: 1` that python.exe is a py virtual env btw

jmlott, 2023-10-06:
It may also be helpful to try running the script on the command line directly by specifying the gcode file after the script. You may get a better error message.

Nathan22211, 2023-10-07:
well I may have found a issue pertaining to this. when running `C:/post_scripts/arc/Scripts/python.exe C:/post_scripts/prusa_slicer_post_processing_script` I get `Usage: python3 ex1.py <filename>` but using python3 mentions downloading the version from the Microsoft store, which reportedly doesn't work for post scripts

Nathan22211, 2023-10-07:
tried the script in prusa and it worked fine, it seems the script doesn't work in Orca Slicer, like at all, I'll need to look at it a bit

foreachthing, 2023-10-07:
> `C:/post_scripts/arc/Scripts/python.exe C:/post_scripts/prusa_slicer_post_processing_script` I get `Usage: python3 ex1.py <filename>`

Shouldn't you be using backslashes and file extensions? If you don't use absolute paths, your script needs to be in OrcaSlicer's directory.

Nathan22211, 2023-10-07:
@foreachthing you'd have to on linux but I don't think windows cares what slash you use. And so far it has worked up to failing completely on orca, I suspect it's an issue with the script itself

foreachthing, 2023-10-07:
I can't test your script on the cell, I'll have a look next week or so...

foreachthing, 2023-10-11:
So, I tested my script with OrcaSlicer 1.7 and it ran without issues.

I ran the script directly from `C:\Users\user\AppData\Local\Temp\orcaslicer_model\Wed_Oct_11\13_10_13#21104#55\Metadata` and then from OrcaSlicer. In both cases, no issues.

my command in OrcaSlicer: `"C:\Users\user\AppData\Local\Programs\Python\Python311\python.exe" "h:\dev\Slic3rPostProcessing\Slic3rPostProcessor.py" -f ;`

Nathan22211, 2023-10-11:
Yeah apparently the arc overhangs script doesn't work in orca and that's what I was using

foreachthing, 2023-10-11:
> Yeah apparently the arc overhangs script doesn't work in orca and that's what I was using

Got a link?

Nathan22211, 2023-10-11:
https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration

foreachthing, 2023-10-13:
@Nathan22211 , I guess you could close here.
Since this is an "issue" with the script, not OrcaSlicer -> https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/issues/58

closed 2023-10-13 by Nathan22211