OrcaSlicer/OrcaSlicer #2334: Missing m109 gcode - Print starts without fully heating extruder

A2-Dev, 2023-10-06:
### OrcaSlicer Version

OrcaSlicer 1.7.0

### OS version

Win 10 Ent

### Additional system information

CPU: Intel Core i7-8700K CPU @ 3.70GHz
Mem: 128GB Dual-Channel Unknown @ 1334MHz (19-19-19-43)
GFX: [ASUS TUF Gaming NVIDIA GeForce RTX 3080 Ti OC Edition Graphics Card - PCIe 4.0 - 12GB GDDR6X

Good 'nuff

### Printer

QIDI X-Max

### How to reproduce

generate any gcode using selected printer, profile was 'Qidi Generic ASA'

### Actual results

generated gcode has no m109 only m104 and then it starts printing

### Expected results

for the extruder to be hot enough before it starts printing?

### Project file & Debug log uploads

[stealthburner.zip](https://github.com/SoftFever/OrcaSlicer/files/12825997/stealthburner.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2023-10-07:
@A2-Dev
This is most likely not related to OrcaSlicer but rather to Marlin itself.
Did you by any chance enable **`ADVANCED_OK`**?

Had the same issue after updating Marlin.

rondlh, 2023-10-18:
I have the same issue, there is a M104, but no M109, which causes the print to start before the nozzle is at the required temperature. I use Marlin 2 G-code with Orca slicer 1.7.

discip, 2023-10-18:
Have you checked your start g-code?

A2-Dev, 2023-10-18:
Here is the unedited start gcode
I believe the gcode in question is coming from dll, but i didn't do much checking.
I am using bambu slicer in the mean time, have not looked at prusa slicer.

`G28
G92 E0
G0 X5 Y5 Z0.3 F3600`

discip, 2023-10-18:
Try this:
```g-code
G28
G92 E0
G0 X5 Y5 Z0.3 F3600
M140 S[hot_plate_temp_initial_layer]		; set bed temp
M109 S[nozzle_temperature_initial_layer]		; wait for nozzle to reach temp
M190 S[hot_plate_temp_initial_layer]		; wait for bed temp
```

rondlh, 2023-10-19:
My start G-code is empty, I guess adding these codes will solve the issue, still it's surprising it's needed, thanks!

discip, 2023-10-19:
@rondlh
No, you would need to add only:
```g-code
M140 S[hot_plate_temp_initial_layer]		; set bed temp
M109 S[nozzle_temperature_initial_layer]		; wait for nozzle to reach temp
M190 S[hot_plate_temp_initial_layer]		; wait for bed temp
```

A2-Dev, 2023-10-19:
Just adding the m109 line is a way to solve the issue, but i believe it's still a bug.
Also shouldn't the m190 come before the nozzle so it isn't just sitting there at full temp waiting for the bed to heat?

Regardless, thanks for the help rondlh, i will use your fix the next time i print. ;)

discip, 2023-10-19:
You are welcome.
Please don't forget to close after this gets fixed one day. 😊

rondlh, 2023-10-19:
Just to be clear...
All my "Machine G-code" boxes are empty (except for comments). 
The generated G-code has "M190 S75 ; set bed temperature and wait for it to be reached."
The added M109 comes after that, which is correct.
Then all works fine for me.

Ftown1404, 2023-11-10:
This was driving me nuts! This is still an issue with 1.8.0 beta 2

wrenchonline, 2023-11-26:
the same problem

closed 2024-03-04 by github-actions[bot]

nicocarbone, 2026-03-05:
This is still an issue in 2.3.1. I believe that M104 S[first_layer_temperature]; should be M109 S[first_layer_temperature];