OrcaSlicer/OrcaSlicer #2339: Between extrusion role change G-Code (Dynamic PA macro)

kubkadl, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
No

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

**Describe the solution you'd like**
Add custom between extrusion role change g-code like it is in SS

**Additional context**
I would like to add this custom g-code to use macros for dynamic PA (https://github.com/hadesz/v2blorgr-config/wiki/Dynamic-Pressure-Advance)

Ocraftyone, 2023-10-07:
That's a really cool idea! Seen as how SS already implemented that, it's just a matter of finding the important parts in SS and transferring them to orca. I'll take a look into doing this when I get a chance.

kubkadl, 2023-10-17:
> That's a really cool idea! Seen as how SS already implemented that, it's just a matter of finding the important parts in SS and transferring them to orca. I'll take a look into doing this when I get a chance.

Have you watched it yet?

Ocraftyone, 2023-10-19:
Yup. Just took a look at it and implementation was pretty simple. I am building a test build and if everything goes smooth, I'll submit a PR

PR #2459 (2023-11-09): SS port - Change extrusion role gcode
Gcode that is run every time that the extrusion role is changed. Is helpful for those wanting to use dynamic PA scripts. Provides values for extrusion_role, last_extrusion_role, layer_num, and layer_z.

fixes #2339