OrcaSlicer/OrcaSlicer #2340: Adaptive Meshing

discip, 2023-10-06:
I searched but couldn't find if there was already a **Feature Request** on this topic.

So I'm proposing to add something like this to OrcaSlicer: **[KAMP](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging)**

Ocraftyone, 2023-10-06:
It is possible that could be implemented, but what is the benefit of implementing it into the slicer if KAMP already exists. If you are using klipper, best practice is usually to use a start print macro in the slicer's custom gcode and to handle all of the start of the print (set temps, home printer, init bed mesh, purge nozzle).

discip, 2023-10-06:
@Ocraftyone
Thank you for replying.
I managed to solve this.

closed 2023-10-06 by discip