OrcaSlicer/OrcaSlicer #2343: Change type doesn't appear in right click menu

PhilBaz, 2023-10-06:
### OrcaSlicer Version

current 

### OS version

win 10

### Additional system information

doesn't matter

### Printer

Voron 2.4

### How to reproduce

When adding a part as a 'negative' you get the 'change part type' menu item to change its type. This should be on the main right click menu. I can't seem to find it anywhere except the part created 'negative'

[image]
[image]

### Actual results

no otpion

### Expected results

Should be able to access this function multiple ways.

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

Ocraftyone, 2023-10-07:
It looks like that is the expected result. You are selecting the entire object in the first picture and an individual part in the second. There are no different types of objects, but there are different types of parts.

PhilBaz, 2023-10-07:
@Ocraftyone 

Ahh, I see. It looks like the 'negative' object has to be created under an existing object like a parenting system. And it doesn't affect other objects. 

This is less flexible than a generalized system. But yeah this isn't a bug, it's just not nicely implemented, by bambu or prusa i imagine.

closed 2023-10-07 by PhilBaz

Ocraftyone, 2023-10-07:
Yeah. I could see the benefit of having a sort global modifier, but many of the use cases could also be accomplished by converting multiple objects to multiple parts under a single object, then adding a modifier to that. And using the one object multi part system, you have to deliberately select which models you want to be modifiable, which could prevent mistakes.