OrcaSlicer/OrcaSlicer #2344: Slicing max flow calibration produces strange results with 0.6mm nozzle

bitingsock, 2023-10-06:
### OrcaSlicer Version

1.7.0

### OS version

Win11

### Printer

Custom Klipper

### How to reproduce

1: configure a printer with 0.6mm nozzle (and appropriate line widths), or use a preconfigured profile that uses 0.6mm (I tried several)
2: start max volumetric flow calibration
3: Slice

### Actual results

The produced gcode contains a bottom infill layer and OS reports `Empty layers around bottom are replaced by nearest normal layers`
[image]

If the bottom layers are then reduced (from the calibration's default value of 1) to 0 and then resliced it almost fixes it but then it refuses to produce an internal brim
[image]

### Expected results

A gcode similar when sliced with a 0.4mm nozzle. A small external+internal brim and no `Empty layers around bottom are replaced by nearest normal layers` error.

### Project file & Debug log uploads

[debug_0layer.log](https://github.com/SoftFever/OrcaSlicer/files/12835152/debug_0layer.log)
[debug_1layer.log](https://github.com/SoftFever/OrcaSlicer/files/12835153/debug_1layer.log)
[Max volumetric speed test.zip](https://github.com/SoftFever/OrcaSlicer/files/12835154/Max.volumetric.speed.test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bitingsock, 2023-10-06:
With some testing it seems to fix itself by:
1: changing the `default` and `outer wall` line widths down any(?) amount
2: slice
3: change line widths back up

the bug seems to be triggered by turning on vase mode while `outer wall` is set >1

Noisyfox, 2023-10-09:
It's not really caused by the vase mode, but rather by the fact that the model's shell thickness is less than the specified wall line width. You cannot print a wall that's are thinner than the extrusion width, hens the slicer thought the layers are invalid.

bitingsock, 2023-10-09:
Ok, so the model has 1mm wall and It slices right with 1.6mm but not 1.7
If this is really the problem then the max flow calibration model might need to be fixed.

Noisyfox, 2023-10-09:
> Ok, so the (part1) model has 1mm wall and It slices right with 1.6mm but not 1.7

Yeah it's slightly more complicated than just the thickness, but that's the major part. I'm still trying to figure out what exactly went wrong, as things like elephant foot could also contribute to the issue to some degree.

bitingsock, 2023-10-09:
The issue realistically only presented itself with the max flow calibration. Perhaps the model should just be solid and sliced with 1 wall, 0 infill, 0 top, 0 bottom.
Still prevents inner brims but that is a preferable outcome over the extra layer.

Noisyfox, 2023-10-09:
The proper solution is to use Arachne.

bitingsock, 2023-10-09:
Yes but only to a reasonable extent. As per my 3mf, I am using Arachne. It is only tuned to compensate so much by default.

Noisyfox, 2023-10-09:
> Yes but only to a reasonable extent. As per my 3mf, I am using Arachne. It is only tuned to compensate so much by default.

It should be good enough to perform the test I think.

bitingsock, 2023-10-09:
The test "works" but is still producing improper slice.

Noisyfox, 2023-10-09:
> The test "works" but is still producing improper slice.

It's "proper" as long as the test can be printed and achieve the goal of calibration. It doesn't need to be perfect.
Though for normal prints, we might need to improve the error message to clearly state the wall width is too thick for the print, and let the user to decide if proceed or change the parameter.

Noisyfox, 2023-10-09:
Ah wait... I get what you mean. Arachne doesn't really solve the problem in this case. I thought it will because I'm running a modified version which I made some changes that (kinda) solve the problem. I'll do more tests.

bitingsock, 2023-10-09:
> Though for normal prints, we might need to improve the error message to clearly state the wall width is too thick for the print, and let the user to decide if proceed or change the parameter.

I'd be fine with that, although the calibration script is still making such a scenario.

> It's "proper" as long as the test can be printed and achieve the goal of calibration (which I'm not sure if it does since I don't have a 0.6mm nozzle printer). It doesn't need to be perfect.

From my print and looking at the gcode (the script is seemingly only injecting G1 FXXXX at each layer) it achieves its goal.

Noisyfox, 2023-10-09:
It was actually the `fix_slicing_errors` that mess the things up. I have it disabled on my local build and things looks much better.

Noisyfox, 2023-10-09:
https://github.com/SoftFever/OrcaSlicer/blob/d54e5bd9d25ef117e2ca2f3288210181fe5865a2/src/libslic3r/PrintObjectSlice.cpp#L671-L673

This part will always offset the layer using the outer wall width, ignore things like variable line width and detect-thin-wall etc., which causes the problem.

Noisyfox, 2023-10-09:
In my opinion, this `fix_slicing_errors` causes more errors than it claims to 'fix', we should just remove this part, just like what PrusaSlicer did a few months ago.

bitingsock, 2023-10-09:
Separate issue but, could it also be a potential culprit of layers that are removed entirely?

Noisyfox, 2023-10-09:
> Separate issue but, could it also be a potential culprit of layers that are removed entirely?

Exactly.

PR #2370 (2023-10-12): Remove `fix_slicing_errors` because it makes no sense
this also fixes #2344 and fixes #2350
commit fc0067cc (2023-10-12): Remove `fix_slicing_errors` because it makes no sense (#2370)

this also fixes #2344
closed 2023-10-12 by SoftFever