OrcaSlicer/OrcaSlicer #2345: Creality K1 won't save Filaments

The760, 2023-10-06:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Printer

Creality K1

### How to reproduce

1. Add Creality K1 as a printer
2. You'll see that only "Default Filament" is available
3. Edit this and save it as a new Filament
4. Close Orca and reopen. The Filament you saved will be gone, and you'll be left with "Default Filament"

### Actual results

[image]

### Expected results

You should be able to save filament profiles as normal, and Orca should retain them. For whatever reason, this does not happen with the K1.

### Project file & Debug log uploads

No project file or log for this.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ocraftyone, 2023-10-07:
I have seen a similar issue like this in superslicer when the filament and printer are not properly marked as compatible with each other.

timmeade, 2023-10-13:
I have hit this issue also if anyone has a workaround .   K1 MAX

SoftFever, 2023-10-13:
@The760 @Ocraftyone 
Can I check if you created the custome filament based on "Default filament"?
If so, please try to select filament for Creality printer first.

timmeade, 2023-10-13:
I was trying to Import the ones from the defaults.  They never show up.

Ocraftyone, 2023-10-13:
Honestly, I'm not sure. What caused it for me was importing a prusaslicer config into superslicer. I just solved it by manually entering the config into SS.

Ocraftyone, 2023-10-13:
Hmm. It should have asked you during the setup process if you wanted to add any filament types.

PR #2767 (2023-11-17): Fix creality k1/max default materials (#2609)
This should fix #2609
Fix #2345
closed 2023-11-17 by SoftFever