OrcaSlicer/OrcaSlicer #2347: Scale flow rate chits (X & Y) on small build plates

jamincollins, 2023-10-07:
**Is your feature request related to a problem? Please describe.**

I'm going through filament calibration on a Voron v0.  I'm at the flow rate.  However, due to the size of the chits created they won't all fit on the print bed.  AFAICT, a simple solution is scaling X & Y down to ~80% of the original size.  I'm guessing this won't impact the calibration (if it will please advise). 

**Which printers will be beneficial to this feature**
Any printer with a smaller build plate than the generated chits.

**Describe the solution you'd like**
See above.

Ocraftyone, 2023-10-07:
Duplicate of #2256

ichibana, 2024-01-04:
@jamincollins i'm in the same boat. did you scale and calibrate without issues?

jamincollins, 2024-01-04:
Yes, just don't touch the z axis

ChaosBlades, 2024-03-11:
Visual representation of issue. It is super close to fitting. Should be a relatively easy no compromises fix. I propose making them taller and narrower. Could also close spacing to 1mm.
[image]

jamincollins, 2024-03-11:
The second set is worse since it has ten tiles.

ChaosBlades, 2024-03-11:
> The second set is worse since it has ten tiles.

Forgot about that.
[image]

 Narrower and shave off some of the space above and below the numbers. Even at 80% scale still plenty of space could be saved.
 [image]

Ocraftyone, 2024-03-11:
I have a model that is similar to the super slicer version. I will try and implement that this week as an option.

closed 2024-05-18 by jamincollins