OrcaSlicer/OrcaSlicer #2350: Max flow rate calibration on MacOS - SpeedTestStructure.step error

DTDakin, 2023-10-07:
### OrcaSlicer Version

1.7.0

### OS version

MacOS 13.5.2 (22G91)

### Additional system information

1.4 GHz Quad-Core Intel Core i5
Intel Iris Plus Graphics 645 1536 MB
Memory 16 GB 2133 MHz LPDDR3

### Printer

BBL X1C

### How to reproduce

1. BambuLab X1 Carbon, 0.6 mm nozzle
2. Select Calibration/More/Max flowrate
3. Error pops up 
4. Since I am using a Mac using the Windows OrcaSlicer the recommended Tip is not an option. So slice anyway.
5. Warning pops up
[image]

### Actual results

1. Error message
2. Warning message
3. First layer prints okay
4. Subsequent layers nozzle is too high and plastic drops to previous layer. Causes calibration to be unuseable.

### Expected results

1. SpeedTestStructure.step should load without errors
2. MacOS version of OrcaSlicer should allow non-manifold fix

### Project file & Debug log uploads

[LogFiles.zip](https://github.com/SoftFever/OrcaSlicer/files/12838077/LogFiles.zip)
[Max volumetric speed test.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12838078/Max.volumetric.speed.test.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-10-08:
Just ignore those warnings, it should slice just fine.

Noisyfox, 2023-10-08:
This seems to be the same issue as #2344

PR #2370 (2023-10-12): Remove `fix_slicing_errors` because it makes no sense
this also fixes #2344 and fixes #2350
closed 2023-10-12 by SoftFever