OrcaSlicer/OrcaSlicer #2353: Supports wont turn off?

borris345, 2023-10-07:
### OrcaSlicer Version

1.7.0

### OS version

windows 11

### Additional system information

2080ti
i9 13900k
64gb ddr5 ram

### Printer

bambu x1 carbon

### How to reproduce

turn supports off still there.
turn supports on there obviously
turn supports off still there

support painting removed them

### Actual results

supports stay on

### Expected results

supports to turn off

### Project file & Debug log uploads

https://pasteboard.co/fEwXP2Fa1FYP.png
https://pasteboard.co/B9WmJgtquF5E.png

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-10-07:
Check your per-obj settings

borris345, 2023-10-08:
What am I looking for that I may have accidentally changed?

Noisyfox, 2023-10-08:
[image]
Check if you have per-obj support turned on.

borris345, 2023-10-09:
cheers that was the prob. there was a lock on it. thanks

closed 2023-10-09 by borris345