OrcaSlicer/OrcaSlicer #2354: Update Filename Format to pass forward slashes

poppinfresh69, 2023-10-07:
Want to send sliced files to specific directories in Klipper (printer 1, printer2, etc.). Manually adding the directory information ie. /E3V2/some_file_name works great in the Print window. The problem is that you have to manually change the directory every time you change printers as it does not tie to a print profile. Trying to add the directory information in the Filename format window is ignored as it does not appear to like the forward slashes.

**Which printers will be beneficial to this feature**
Klipper

**Describe the solution you'd like**
Being able to add the directory information in the Filename Format window and having it pass to the file name would fix this issue as it would be tied to what is saved in the filament profile. Changing filament profiles would also change the directory info.

**Describe alternatives you've considered**
If the directory info could update automatic from the printer profile it would be ideal, but my proposed seems like the quickest and easiest fix for now.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

Ocraftyone, 2023-10-13:
I'll look into this when I get a chance. Doesn't seem like it would be too hard to implement.

closed 2024-01-20 by github-actions[bot]

Eldenroot, 2024-01-20:
Still relevant

outsolder, 2026-10-08:
yup, still failing :)