OrcaSlicer/OrcaSlicer #2356: Improve multicolour on a single extruder machine

cpasoft, 2023-10-08:
Many of us use Orca with multiple filaments on single extruder printers. Typically, we would edit the G code for the filament change with a pause code and then perform the manual filament change.

However, we have a problem because Orca adds an M600 command after the initial heating phase, together with the T0, T1...Tx tool change codes, which we do not need as we only have one extruder.

[image]

[image]

[image]

Therefore, I suggest that when you check the single extruder option, OrcaSlicer automatically prevent the generation of the initial M600 command and the tool change codes (T0, T1...Tx). This is important because on some machines it causes errors that render them non-functional. For example, it causes problems with Bambu printers without AMS.

[image]

Sorry, the screenshot is in Spanish, but it says something like "Could not get the AMS mapping table; click retry to continue".

cpasoft, 2023-10-08:
*Note: For the first M600 there is a workaround...
You must write on the "Change filament G-Code" section:
`{if layer_num > -1}M600{endif}`
instead of a single M600.

There is not workaround for the Tx codes right now.

cpasoft, 2023-10-08:
For the second problem, I've written a little script in Windows to solve it.

Just create a Remove_T.cmd (or something similar) and then call it from the postprocessing script as:
"C:\wherever it is\Remove_T.cmd"

<@echo off
if "%~1"=="" (
  echo Uso: %0 archivo
  exit /b 1
)

findstr /V /R /C:"^T[0-9]" "%~1" > "%temp%\temp.txt"
move "%temp%\temp.txt" "%~1" > nul
>

Anyway, It should be solved by Orca itself.

PegasusSeiya12, 2023-10-08:
It would be a notable improvement and an advantage for this laminator over the competition.

waxall, 2023-10-10:
I am using my single toolhead klipper machine with a PAUSE command to swap filaments but when when I RESUME the print, I get an error like "Extrude only move too long ( 92233720368547.766mm vs 200.000mm)" I've also set my M83 and G92 E0 commands but that error keeps being thrown up. 

Was wondering if this is related to the T0 issue

closed 2023-10-22 by SoftFever

cpasoft, 2023-10-23:
Tested and working!!!...

Thank you all... :-)

PR #2379 (2023-10-22): Add "Manual filament change" option and logic
This option does a few things that make a SEMM setup where you are manually swapping the filament on pauses a bit easier

1) disables calling the "on filament change" gcode during setup. This may be useful on a machine that can change the filament itself, but for manually changing the filament, it is an unnecessary pause. It should be assumed the first inserted filament is the first one to be used.

2) tool change command is changed to a gcode tag. This prevents errors on machines that do not understand the command possibly resulting in a failed pr [... cut]