OrcaSlicer/OrcaSlicer #2357: Material change as first layer ?

guestisp, 2023-10-08:
I think would be useful to thread the first layer after a material change like a first layer (slow down, higher temp, and so on). This help the adesion a lot if you print with different materials.

Actually, in PrusaSlicer, I have to add a tons of layer customization immediatly after the M600.

Adding something that handle this automatically, won't be useful ? Somethign like a flag in the filament settings....

guestisp, 2023-10-08:
Or a generic modifier that applies all the customization we have for the first layer, at a specific layer, in example, you add a customizer at layer 85 with a flag: "Handle this as first layer" true/false

blackhead1981, 2023-10-13:
If I understood you correctly, there may be a solution for this:
go to "printer settings" - "machine g-code" - "change filament g-code"
and insert your own g-code

guestisp, 2023-10-13:
yes but you have to set A LOT of parameters, manually, like all speeds, temperatures and so on

closed 2024-01-20 by github-actions[bot]