OrcaSlicer/OrcaSlicer #2358: Print order for the first layer

guestisp, 2023-10-08:
I've asked the same months ago on superslicer (or prusaslicer, don't remember)
I don't see any single reason why, on first layer (the one attached to the bed, obviously) printing inner -> outer would be better than printing outer -> inner.

Usually, on first layer you want the max adhesion, and you can get this priting longer lines. printing from the inner lines are obviously shorter than printing from the outer (in examples, holes: an M3 hole is more stable if printed from the outside)

So, why not adding a flag to print external perimeters first on first layer ?

closed 2024-02-12 by github-actions[bot]

trebory6, 2024-10-26:
Ok, am i taking crazy pills or is this common sense?

I'm having first layer adhesion issues, and the parts that print outer perimeter -> inner perimeter on the first layer have a 100% success rate over the parts that print inner perimeter -> outer perimeter. 

Unfortunately there is no way to control this manually. So I literally have to slice my plate with multiple objects, then print it then cancle the objects that print inner -> outer. 

If I delete the ones that print inner -> outer before I slice it, it'll just randomly select some other objects to print inner -> outer. It's completely random for some reason.

TeaRex-coder, 2025-06-21:
+1 this should be a thing