OrcaSlicer/OrcaSlicer #2359: Unwanted extrusion after wipe tower

wiggywoosaa, 2023-10-08:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11

### Additional system information

CPU : 12gen i9

### Printer

RatRat

### How to reproduce

Once sliced there is an additional unwanted extrusion command when starting the layer after moving off of the wipe tower

### Actual results

The command creats a blob on the start of the layer
[image]

### Expected results

No blob at the start of the layer

### Project file & Debug log uploads

[debug_network_Sat_Oct_07_18_25_04.log](https://github.com/SoftFever/OrcaSlicer/files/12840457/debug_network_Sat_Oct_07_18_25_04.log)
[project.zip](https://github.com/SoftFever/OrcaSlicer/files/12840468/project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

53Aries, 2023-11-06:
Can confirm issue. I am able to work around it by using firmware retraction.

SoftFever, 2023-11-07:
@wiggywoosaa @53Aries 
Can you share a project file again?
For some reason I'm not able to download the attached project.

Meanwhile, can you check if you have set extra de-retraction length in these two places? 
[image]

53Aries, 2023-11-07:
Hello. I will figure out how to upload a file here tonight. I can say
however that i have 0 for both extra length settings if that helps.

SoftFever, 2023-11-07:
> Hello. I will figure out how to upload a file here tonight. I can say however that i have 0 for both extra length settings if that helps.
> […](#)

you can zip the project file and drag & drop here

53Aries, 2023-11-08:
I see where the values are coming from. Line 861 is the unwanted extrusion. It comes after wipe tower operations are complete and the print object has started. .18 is the unretract from line 857(which is based on wipe while retracting) and the 2.0 is from Extruder tab retraction when switching material. 

[image]
[image]
[frog_MMU_4colors.zip](https://github.com/SoftFever/OrcaSlicer/files/13291246/frog_MMU_4colors.zip)

53Aries, 2023-11-08:
Adjusting settings as shown resolves the issue. 
[image]
[image]

SoftFever, 2023-11-09:
@53Aries 
Thank you the information.
I've pushed a fix for it.
The issue should have been fixed in latest nightly build.

SoftFever, 2023-11-18:
Fxied in 1.8

closed 2023-11-18 by SoftFever